OpenPHDGuiding/phd2 #685: PPEC do not invalidate model when guiding stops unless RA has changed

agalasso, 2018-04-19:
We currently invalidate PPEC's model whenever guiding stops.  The model needs to be invalidated since the worm could be out of sync with the PPEC model when guiding resumes.

If we read the RA from the mount when guiding stops and find that RA has not changed when guiding starts again, we could retain the model (same as the current Pause/Resume functionality)

Taking this a step further we could potentially shift the model based on the new RA when guiding starts; this would allow PPEC to retain the model even after a slew in RA.

eklenske, 2018-04-23:
I think that's a good idea! We just have to be careful with the intermediate data points. We probably have to invalidate the "measurements" during the pause, like we do with data during dithering/settling. It probably makes sense anyway to have a "settling" period after a pause, because usually the guide star will drift away a bit.

agalasso, 2018-04-23:
thanks Edgar.

I think I understand what you are saying about the settling period: we could actually treat the stop/start just like a dither and use a dither offset value corresponding to the RA offset (difference in RA between stop and start).

We'd probably want  to have a maximum RA offset and revert to the reset() if the RA offset is more than perhaps 1/2 worm cycle?

agalasso, 2018-07-21:
fixed by f083d2a0aa2f32183e34c14a2677087099d67285

closed 2018-07-21 by agalasso