OpenPHDGuiding/phd2 #686: ZWO cameras: need better persistence of camera selection when multiple ZWO cameras

agalasso, 2018-04-22:
ZWO camera selection persistence uses the ordinal number of the enumerated cameras.  If the enumeration changes then the saved selection is invalid.

PHD2 could use the camera name + ordinal number for the selection.   This would still be a problem if there were multiple cameras with the same name enumerated in a different order, but would be an improvement over what we have now and would cover the common case of different model imaging camera + guide camera.

agalasso, 2018-08-21:
for USB 3 cameras, we may also be able to use ASIGetID() to identify the camera.  needs testing

d33psky, 2018-08-24:
Not sure about Windows and Mac, but the Linux udev system where you can name hardware devices like this : 

# SX-AO
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="FTFE88G0", MODE="0666", SYMLINK+="sx-ao-lf"

# ZWO camera
SUBSYSTEMS=="usb", ATTRS{idVendor}=="04b4", ATTRS{idProduct}=="6572", MODE="0666", SYMLINK+="ZWO_ASI1600MM-Cool"

# ZWO filterwheel
SUBSYSTEMS=="usb", ATTRS{idVendor}=="03c3", ATTRS{idProduct}=="1f01", MODE="0666", SYMLINK+="ZWO_EFW"

might have an equivalent on Windows and Mac with which hardware can be identified and named conveniently.

agalasso, 2018-08-25:
Thanks for the idea.   I think we have what we need from the ASI SDK.  The SDK can give us the model name of each attached camera, and phd2 can connect to the device with the desired model name.

We don't have any solution for the case when there are mare than one camera of the same model.  Even with udev rules, the cameras will have the same USB VID/PID and the cameras do not report a serial number USB attribute (at least not the cameras I have), so we have no way to distinguish multiple cameras of the same model.

commit 692a5589 (2018-08-28): ZWO cameras: use camera model name to decide which camera to connect to. Fixes #686
closed 2018-08-28 by agalasso