OpenPHDGuiding/phd2 #690: RFE: opt out QHY and ZWO camera libraries

mattiaverga, 2018-05-07:
Can you add a way to build PHD2 without linking QHY and ZWO camera libraries?
The reason I ask this is that I'm packaging PHD2 in Fedora (and possibily EPEL) repositories, which don't allow embedding pre-built libraries in packages, so I have to delete contents of cameras/qhyccdlibs and cameras/zwolibs.

With version 2.6.4 I managed to exclude such camera libraries with a [small patch](https://src.fedoraproject.org/rpms/phd2/raw/master/f/phd2_disable_cams.patch), but with the new 2.6.5 version the [build fails while linking the executable](https://copr-be.cloud.fedoraproject.org/results/mattia/Astronomy/fedora-rawhide-x86_64/00750616-phd2/build.log.gz).

Aren't those cameras usable also by INDI drivers?

agalasso, 2018-05-07:
fixed by 0d87a84540e90bb17e6d4183fd32d422c410861f

Linux build instructions have been updated:
https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux

You can now run cmake like this to exclude binary drivers (ZWO and QHY):

cmake -DOPENSOURCE_ONLY=1

agalasso, 2018-05-07:
> Aren't those cameras usable also by INDI drivers?

correct, they are usable via INDI.   INDI probably has to use the binary blob SDKs itself though.

closed 2018-05-07 by agalasso