OpenPHDGuiding/phd2 #704: compile of phdupdate.cpp fails (linux Tumbleweed), with fix

DerPit, 2018-07-07:
Hi, 
just wanted to try phd2 on my linux machine (opensuse Tumbleweed).  Compile of phdupdate.cpp failed with
/home/pit/Sources/phd2/phdupdate.cpp:318:60: error: invalid cast from type ‘const wxString’ to type ‘const char*’
         curl_easy_setopt(curl, CURLOPT_URL, (const char *) url);
(and a second time for installer_url in line 495).  Can be fixed using ..url.char_str() method (patch attached)

[phdupdate.cpp.diff.gz](https://github.com/OpenPHDGuiding/phd2/files/2172802/phdupdate.cpp.diff.gz)

agalasso, 2018-07-07:
thanks for the issue report and for the patch
fixed with c62b9a821543c182b4c6928c02319682f26e1f7c

closed 2018-07-07 by agalasso

DerPit, 2018-08-27:
Maybe rather re-use this thread than opening a new issue:
Same problem again with guider.cpp in v2.6.5dev3-14-gaafdc935:
/usr/local/src/phd2/guider.cpp:122:42: error: invalid cast from type ‘wxString’ to type ‘const char*’
     std::istringstream is((const char *) s);

change 's' to 's.char_str()' fixes it.

agalasso, 2018-08-27:
thanks for letting us know.  it's fixed now (again) d6ae4eefa21f3d87625dd8ed223cc7061eb55d6a