OpenPHDGuiding/phd2 #706: Build is failing on Raspbian

phomer60, 2018-07-09:
I have been building phd2 for a number of years but suddenly I am getting the following errors.
Can anyone suggest the cause of this.

[ 49%] Building CXX object CMakeFiles/phd2.dir/config_indi.cpp.o
/home/pi/phd2/config_indi.cpp: In member function ‘void INDIConfig::UpdateControlStates()’:
/home/pi/phd2/config_indi.cpp:184:27: error: ‘isServerConnected’ was not declared in this scope
     if (isServerConnected())
                           ^
/home/pi/phd2/config_indi.cpp: In member function ‘void INDIConfig::OnConnectButton(wxCommandEvent&)’:
/home/pi/phd2/config_indi.cpp:266:27: error: ‘isServerConnected’ was not declared in this scope
     if (isServerConnected())
                           ^
In file included from /usr/include/wx-3.0/wx/defs.h:695:0,
                 from /usr/include/wx-3.0/wx/wx.h:14,
                 from /home/pi/phd2/phd.h:47,
                 from /home/pi/phd2/config_indi.cpp:39:
/home/pi/phd2/config_indi.cpp: In member function ‘void INDIConfig::Connect()’:
/home/pi/phd2/config_indi.cpp:299:33: error: ‘isServerConnected’ was not declared in this scope
     wxASSERT(!isServerConnected());
                                 ^
/usr/include/wx-3.0/wx/debug.h:280:42: note: in definition of macro ‘wxASSERT_MSG’
             if ( wxTheAssertHandler && !(cond) &&                             \
                                          ^~~~
/home/pi/phd2/config_indi.cpp:299:5: note: in expansion of macro ‘wxASSERT’
     wxASSERT(!isServerConnected());
     ^~~~~~~~
CMakeFiles/phd2.dir/build.make:1849: recipe for target 'CMakeFiles/phd2.dir/config_indi.cpp.o' failed
make[2]: *** [CMakeFiles/phd2.dir/config_indi.cpp.o] Error 1
CMakeFiles/Makefile2:92: recipe for target 'CMakeFiles/phd2.dir/all' failed
make[1]: *** [CMakeFiles/phd2.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

pchev, 2018-07-09:
isServerConnected was added to Indi BaseClient in December 2017, so you need a libindi-dev version more recent than this date.

Best to use the Indi Client provided with phd2, for that remove USE_SYSTEM_LIBINDI from your cmake command.
If this is already the case, run "rm -rf *" in the build directory to force cmake to unpack the new version.

agalasso, 2018-07-11:
Paul, Have you tried what Patrick suggested?

phomer60, 2018-07-11:
Andy, 

Yes, I ended up just doing the rm -rf * and I can now confirm that it worked.

Thanks a lot to Patrick.

Paul

agalasso, 2018-07-11:
ok, thanks for the update. closing the issue.

closed 2018-07-11 by agalasso