OpenPHDGuiding/phd2 #714: request to make bookmarks persistent

agalasso, 2018-08-09:
From Kym:

Is it possible to make the bookmarks persistent?
It would also be nice to be able to save and load them.

agalasso, 2018-08-09:
More info from Kym:

> We are using phd2 on out 1.27m telescope at the University of Tasmania.  
> What we like to do is put our targets on the centre of our imaging CCD.  
> Having the target in the same place makes the flux measurements less sensitive to any flat field errors.
> With bookmarks we can mark the location of the guide star in phd2, so that when we come back to that target we know where the guide star has to be, to have the target in the centre of the science CCD.
> 
> Having bookmarks persistent between phd2 restarts would be good.
> Being able to save and load bookmarks would be great, maybe a simple ascii text file.
> Having a label to identify (Single character) would be excellent.

bwdev01, 2018-08-09:
Is this intended as a further fine-tuning after plate solves?
 
Bruce
 
  _____  

(#714)
 

it on GitHub, or mute
YlgdqPzks5uPDoIgaJpZM4V1Dnj>  the thread.
<https://github.com/notifications/beacon/AMZ5V6Y0OZ4UHPLrjTs3o-kVDEpvSwIEks5
uPDoIgaJpZM4V1Dnj.gif>

closed 2018-08-14 by agalasso