OpenPHDGuiding/phd2 #715: INDI: allow aux mount connection to mount that does not have GUIDER interface

agalasso, 2018-08-12:
PHD2 is excluding any INDI telescope devices that doe not have the GUIDER interface.  It should be accepting devices with the TELESCOPE interface for use as an Aux mount even if they do not have the GUIDER interface.

closed 2018-08-13 by agalasso

DerPit, 2018-10-12:
Is this supposed to work properly?
I do have a synscan mount, handled by INDI.  I can indeed select an INDI Aux mount, and in the config tab the connection to INDI is visible (localhost+port), and I am offered the synscan as mount.  But trying to connect ends in a timeout :( (EKOS is also connected to the mount)
(2.6.5dev5, INDI 1.7.4)

agalasso, 2018-10-12:
@DerPit  could you either open a new issue or post in the phd2 forum. I don't think your problem has anything to do with this issue related to aux mount.  you should just select INDI for the mount , I don't think you need aux mount at all

DerPit, 2018-10-12:
indi_synscan doesn't support pulseguide, that's why I need the ST4 port (plus AUX mount).  It isn't offered as option for normal mount.  That's why I thought it would fit here.  Will open another ticket.  Thanks.

commit 6228f4e5 (2018-08-13): INDI Aux Mount selection should not exclude mounts that do not have the GUIDER interface. Fixes #715