OpenPHDGuiding/phd2 #719: ASI290MM: unreliable subframe readout

DerPit, 2018-08-27:
Hi,
I'm using phd2 with an ASI290MM mini camera on Linux (openSUSE Tumbleweed).
When using it with default settings (subframe mode enabled), after selecting a guide star in loop mode every second image is blank and phd2 reports a lost star.  Switching off subframe mode everything works as expected.

This was with v2.6.5dev3-2-g78ee9f0c.  I'm meanwhile at v2.6.5dev3-14-gaafdc935, but could not yet test it at night.

agalasso, 2018-08-27:
One possibility might be that the ZWO SDK library is somehow not returning the requested sub-frame.

You might want to try connecting the camera to a Windows or Mac PC if you have one available, and/or try using the ASI INDI driver to connect to the camera.  Seeing which of those variations works (if any) might give us a clue where the problem is.

Another idea would be to try increasing or decreasing the size of the search region (brain > Guiding tab) to see if the problem is affected by the size of the sub-frame.

DerPit, 2018-08-30:
Sorry, it took a moment to find time to do more tests.
I only have Linux, but I tried what you suggested, using the INDI connection.  This actually made me realize that most likely I did have EKOS/INDI running when I had the issues with the 'lost' frames.  My main camera is also an ASI (1600MM), so INDI automatically grabs the camera, too when it looks for ASI cameras.  With the test (running phd2 via INDI) I noticed that acquiring a frame with the main cam while phd2 was looping frames from the guide cam made the image acquisition fail repeatedly, only evey 3-4 exposure finished download and display.  So there seem to be some collisions for sure on the INDI/EKOS side.
So I went back to using ASI interface in phd2, left INDI/EKOS running, but disconnected the cam in the INDI configuration, and re-enabled subframes.  At the moment I cannot reproduce the lost frames anymore, but this is daytime, and no star images available.  I'll need to test this at night again and pay more attention to the interplay between guide- and main cam and EKOS/INDI.
I'll report back with my findings - might take some time though.

agalasso, 2019-05-09:
closing this since it was not reproducible and we have not had any additional reports of this problem

closed 2019-05-09 by agalasso