OpenPHDGuiding/phd2 #723: Test if server is receiving commands

jeffpkamp, 2018-09-03:
I am trying to use  the server on 4400 to control phd2.  

I have no problem recieving messages and get all the ones expected, however it doesn't look like any of my messages are getting through.  I have checked, and the server listening on 4300 is working fine.  Can anyone help me trouble shoot this?

This is the program I am running to test the server:

```

import socket, thread
s=socket.socket()
s.connect(("",4400))

def recv(s):
    m=" "
    print s
    s.settimeout(2)
    while m:
        try:
            m=s.recv(1000)
            print m
        except socket.timeout:
            pass

thread.start_new_thread(recv,(s,))

a='{"method": "get_exposure", "id": 1}'
s.sendall(a)
b='{"method": "set_exposure", "params": [1500], "id": 2}'
s.sendall(b)
c='{"method": "set_exposure", "params": [1502], "id": 3}'
s.sendall(c)

```

agalasso, 2018-09-03:
Hi Jeff,  The messages and responses are newline-terminated.  Here's some code to get you started:

```
import socket, re, sys

def wait_response(f):
    while True:
        line = f.readline()
        if not line:
            raise EOFError("PHD2 exited")
        if re.search('"jsonrpc".*,"result":.*,"id":\d+', line):
            return line
        match = re.search('"jsonrpc".*,"error":(.*),"id":(\d+|null)', line)
        if match:
            err = match.group(1)
            sys.stderr.write("PHD2 error: " + err + "\n")
            err = re.sub('.*"message":"(.*)".*', '\1', err)
            raise ValueError(err)

# === main ===                                                                                                                                                                         

s = socket.socket()
s.connect(("",4400))
f = s.makefile("r+", 1);  # line-buffered                                                                                                                                              

print("send get_exposure")
f.write('{"method": "get_exposure", "id": 1}\n')
r = wait_response(f)
print("response = " + r.rstrip())

print("send set_exposure(1500)")
f.write('{"method": "set_exposure", "params": [1500], "id": 2}\n')
r = wait_response(f)
print("response = " + r.rstrip())

print("send set_exposure(1502)")
f.write('{"method": "set_exposure", "params": [1502], "id": 3}\n')
r = wait_response(f)
print("response = " + r.rstrip())
```

jeffpkamp, 2018-09-03:
Ah, the new line character.  That was it.  I figured it was something simple like that.  Thank you.

closed 2018-09-03 by jeffpkamp