OpenPHDGuiding/phd2 #724: Layout in high-resolution is still lacking

shlomiv, 2018-09-09:
Hi,
I just recently started using phd2. I am using it with a new laptop that has high resolution (unfortunately, as I have this problem with many software!). I read in the changelog that starting version 2.64 there were some adjustments of layouts to high-res displays, however some windows are still lacking. 

A good example is the following: 
[image]

A few more are similar, where sometimes I cant read some of the text because it is hidden and there is no option to reveal it, other times the area and font size are so small such that it is very hard to read, especially when outside with the setup. At the very least I would have loved to have all the windows and popups resizeable  and then maybe be able to control the font size..

Thank you all so much for this wonderful application,
Shlomi

agalasso, 2018-09-09:
Shlomi, thanks for the report.  AFAIK none of the PHD2 developers has a high res display so it is going to be difficult for us to systematically go through the app and fix everything. MS Windows does not give us any way to just fix everything automatically, it requires manually going through and fixing each screen one at a time.
If there are specific areas (besides the Polar DriftAlign tool window that you already pointed out), we may be able to fix them if you can let us know where they are. We will give higher priority to screens that are more important to the core functionality of the app, like the main window or the graph.
We may need your help in testing any changes that we make.

shlomiv, 2018-09-09:
Thank you for such a quick reply!

It will be my pleasure. I will go through the windows and any one that I see with a problem I will post a screenshot of. I would also be more than happy to test your changes!

Thanks :)

shlomiv, 2018-09-09:
First, some windows I cant open for the next few weeks because I cannot connect my mount to my computer. (The cable is on its way)

Notes:
1. The bottom bar with the icons (brain, stop, tracking etc), is really small. It would have been awesome if the icons would have been larger.

Screens:
- Tools
    - [ ] Manual Guide..Modify Calibration - Cant open, no mount
    - [X] Adjust lock position - Looks good 
    - [ ]  Comet tracking
[image]
    - [ ]  Star Cross Test - Cant open, no mount
    - [ ]  Guiding Assistant - Some text is clipped
[image]
    - [X] Drift Align - looks good
    - [ ]  Polar Drift align - see first post
    - [ ]  Static Polar Alignment - Not sure if this is good or too condensed. There are two pictures, one with the instructions which are so small and unreadable (and lots of scrolling to do..)
[image]
[image]
- Darks
    - [X] Bad Pixel Map - Looks good
    - [X] Import from profile - Looks good   
- Help
    - [ ] About - Although not critical, its really small and hard to read :)
[image]
    - [ ] Upload log files - not sure if there is something missing as the next button is disabled
[image]

Here is a picture of the entire screen to see the ratios. The titles of each pane seems a bit clipped. Not critical, but not aesthetic. 
[image]

Hope this helps! 

Shana Tova to all, 
Shlomi

agalasso, 2018-09-09:
Thanks, that is extremely helpful. We will update the ticket when we make some progress.

agalasso, 2019-02-13:
see also #551 (duplicate)