OpenPHDGuiding/phd2 #725: feature request: shorter exposure time for dither recovery

agalasso, 2018-09-18:
Message from Tim: https://groups.google.com/d/topic/open-phd-guiding/tiaLB9jXDFo/discussion

> As mounts continue to get better and particularly with the increasing proliferation of encoder mounts, longer guide exposures start to have more and more value. In my case (iOptron CEM120EC observatory-mounted) I have very little RA PE and excellent polar alignment. So guiding is only really necessary to correct for refraction, mount flexure and very small PAE. I find that guide camera exposures of 15 or more seconds all but eliminate erroneous corrections due to seeing and my results are generally very good. The only problem with this approach is dithering - or more precisely - settling after a dither. With 15 second exposures, it can take a lot of precious imaging time to settle after a dither. Have you gents given any thought to offering an option to shorten the exposure time during settling?

agalasso, 2018-09-18:
related issue: #404

agalasso, 2020-01-27:
The tricky part of implementing this is to determine an appropriately short the exposure time that will not lose the guide star.  This would need to be done dynamically (similar to how auto-exposure mode works) to support automated systems where guide stars with varying SNR are selected over the course of an automated imaging session.

bwdev01, 2021-05-16:
A related change would be to ignore the "time lapse" setting during settling.  In lieu of using long cam exposures, some users with very accurate mounts use time-lapse.  This approach would avoid the difficulties of losing guide stars.

bwdev01, 2022-09-11:
Implemented via the 'variable delay' feature in 2.6.11dev1

closed 2022-09-11 by bwdev01