OpenPHDGuiding/phd2 #726: request option to minimize the guiding assistant window

agalasso, 2018-09-28:
from Michael:

> Would it be possible to add the ability to minimize the Guiding Assistant window, the better to see what's going on in the graphical display?

bwdev01, 2018-09-28:
Use-case description from OP:  Screen real estate is limited and he wanted to look at the guiding graph to see the range of PE excursion.

bwdev01, 2018-09-28:
The use-case can be handled by un-docking the guiding graph.  Longer term, we already have an item to rework the GA layout and make its window resizable.  I think that's a better approach.  When the GA is running, all important controls and status messages are in that window - the user needs to focus on those.  If someone inadvertently minimizes or hides the GA window, it will be "lost" and chaos will result - just not a good UI design IMO.

closed 2018-09-28 by bwdev01