OpenPHDGuiding/phd2 #727: have the option to use several dark libraries

Hansi123456789, 2018-09-29:
Hi, some people may want to use different camera settings for their guider in certain situations, or one may want to use different dark libraries for different temperatures ( say one for winter, te other for summer)
Currently, phd guiding only supports one single library, which then has to be renewes whenever one changes the gain or resolution of the guiding camera...

the master dark is probably a single file. can you make it possible that phd enables one to create several masterdarks for different guider settings and that the user may choose a specific masterdark for certain temperatures..

agalasso, 2018-09-29:
You can make a separate equipment profile for each dark library.  Would that work for you?

agalasso, 2018-10-18:
no response from @Hansi123456789, assuming issue is resolved by prior comment

closed 2018-10-18 by agalasso