OpenPHDGuiding/phd2 #728: Aux Mount: Cannot connect to mount, coredumps when deselecting

DerPit, 2018-10-12:
Hi,
I'm running PHD2 on a dedicated small Linux PC next to the mount that also runs an INDI server to manage the gear.  The mount is a GP-DX with EQ5 upgrade kit, controlled by indi_synscan.  As this driver doesn't support pulse guide I cannot use it as (main) mount.  So I wanted to use the Aux Mount feature for this.
I can select 'INDI mount' for aux, then open the config tab.  This has already connected to the local indi server, and I can select 'synscan' as mount and hit 'OK'
If I then try to connect to the mount nothing happens, and after some time I get a timeout error.
It wasn't obvious if I have to enter the device (port), too (documentation suggested for ASCOM you have to do this).  So I entered it, but it did not change:  No connection.  However, when I now want to deselect the Aux Mount (set back to 'None'), phd2 crashes with a coredump:
free(): invalid pointer
Aborted (core dumped)
phd2 2.6.5dev5 / INDI 1.7.4 / EKOS 2.9.8

agalasso, 2018-10-12:
re: hang: could you attach the phd2 debug log from the session; a debug log with PHD2 INDI verbose logging enabled would be even better
re: core dump:  could you provide the stack backtrace from the coredump
thanks

pchev, 2018-10-12:
Andy, is AUX mount using the same scope_indi.cpp ?
In this case if the driver do not send TELESCOPE_MOTION or TELESCOPE_TIMED_GUIDE properties the CheckState() function never set Ready.
We must have a way to know if the driver is use as a AUX to only check for EQUATORIAL_COORD property.

agalasso, 2018-10-12:
Patrick, you're right, that is the problem!
I'll push a fix shortly.

agalasso, 2018-10-12:
@DerPit, I pushed a fix but I was only able to test with simulators. Could you please pull the latest code and give it a try?

DerPit, 2018-10-12:
Yeah!  It now indeed nicely connects to the mount.  The rest has to wait a bit, as the equipment is in use at the moment for imaging ;^>

DerPit, 2018-10-13:
Yes, communication with the mount works, it does now see the proper coordinates.
I tried doing a meridian flip calibration, but that doesn't work, it cannot detect the pier side, and therefore the 'Calibrate' button stays greyed out.  
Manual flip of calibration does work (calibration at equator on one pier side, then flip and track something  at high declination). 
So I assume this issue can be closed?  Or should I try to look at that coredump?

agalasso, 2018-10-13:
great, glad to hear it is working now.  No need to bother with the core dump, it was a side-effect of the issue that was fixed.
The missing side of pier info is a limitation of the INDI driver and would need to be fixed in the INDI driver.

agalasso, 2018-10-13:
fixed with abde0c6ea65d5678a9e6c989013df1b1d718dc11

closed 2018-10-13 by agalasso

DerPit, 2018-10-16:
Sorry to bother again :(
The pier side info is available in the synscan driver, and I can even see it in the INDI browser of PHD2
(Connect->Aux Mount->Config->Other Options: INDI) under Mount Information
[image]
But connecting or trying to run the pier calibration tool does produce the log entry
16:35:32.600 00.165 140092500392512 Throw from /usr/local/src/phd2/scope_indi.cpp:966->INDI Scope: not capable of getting side of pier
16:35:32.601 00.001 140092500392512 INDI Mount: SideOfPier returns -1

agalasso, 2018-10-16:
could you attach a phd2 debug log with PHD2 [INDI verbose logging](https://groups.google.com/forum/#!msg/open-phd-guiding/W9k20OpAlts/-ZQkaKOVAQAJ) enabled

DerPit, 2018-10-17:
Here it goes.  I just fired up, connected devices, and opened the Pier side calibration up to the second screen.
[PHD2_DebugLog_2018-10-17_172634.txt.gz](https://github.com/OpenPHDGuiding/phd2/files/2488358/PHD2_DebugLog_2018-10-17_172634.txt.gz)

agalasso, 2018-10-17:
Thanks, that was the info we needed.   The INDI driver is not sending the property TELESCOPE_PIER_SIDE.   Apparently it has the information available in the driver since it is being displayed in the mount information section.  The problem is that it is not being sent as the standard INDI TELESCOPE_PIER_SIDE switch property.
You'll need to contact the INDI driver developer and see if they can fix it.  I don't expect it would be hard for them to fix since it seems the driver has the information available and is just not sending it.

DerPit, 2018-10-18:
Jasem just committed a patch to the indi git, now also pier side reporting (and automatic calibration switch) do work with the synscan aux mount :)