OpenPHDGuiding/phd2 #729: exclude device SynScan not a valid mount

starhopper-de, 2018-10-16:
Hello Forum,
I have tried to connect my Skywatcher EQ8 with PHD via INDI. The mount connects to the INDIlib, but I cannot manage to connect to PHD. The log said: "it is not a valid mount". The Syncan firmware is new, the indilib also. It works under ASCOM. 

What can I do? I do not think, that it is PHD issue. I asked this in the indi forum also.

Thomas

PHD log snipplet
21:40:25.628 00.001 140068463302400 INDIConfig: newDevice SynScan
21:40:25.628 00.000 140068463302400 device SynScan interface(s) TELESCOPE
21:40:25.629 00.001 140068463302400 exclude device SynScan not a valid mount

agalasso, 2018-10-16:
Thomas, we'll need to see the full debug log, preferably a log with [PHD2 INDI verbose logging](https://groups.google.com/forum/#!msg/open-phd-guiding/W9k20OpAlts/-ZQkaKOVAQAJ) enabled.

starhopper-de, 2018-10-17:
Hello,
I hope I got the right logs. Thanks for watching.
Thomas

[image]

[indi_synscan_telescope_15h38m56.log](https://github.com/OpenPHDGuiding/phd2/files/2488269/indi_synscan_telescope_15h38m56.log)
[PHD2_DebugLog_2018-10-17_174557.txt](https://github.com/OpenPHDGuiding/phd2/files/2488270/PHD2_DebugLog_2018-10-17_174557.txt)

agalasso, 2018-10-17:
Thomas, could you try again with the "Verbose logging" option checked in the INDI Mount Selection window and attach the PHD2 debug log from that session.
Thanks

pchev, 2018-10-17:
The problem is because the indi_synscan_telescope driver do not support guide command:
SynScan.DRIVER_INFO.DRIVER_INTERFACE=1
The value must be 5 if guide command are supported.
So the message given by PHD2 is perfectly valid. 

Maybe the message can be changed to "exclude device SynScan not a valid mount with pulse guiding" to be more clear.

Thomas, you cannot use this driver for guiding. It as only the capability to slew the mount and report it's coordinates. This can probably be implemented but this require to work on the INDI driver itself.

So at the moment you can only use it as AUX mount ( if you compile the last PHD2 code that fix #728 ), with ST4 guiding throughout the camera.

Or use indi_eqmod_telescope that work perfectly with PHD2.

Patrick

starhopper-de, 2018-10-18:
Thank you very much for clarification. I will tell this to Jasem, maybe he can put this information to the driver page. I will look for a EQDirect cable and use EQMOD.
Many thanks - Thomas

agalasso, 2018-10-18:
closing as it is not a PHD2 issue

closed 2018-10-18 by agalasso