OpenPHDGuiding/phd2 #734: Documentation: Overview about available "Polar alignment" methods in PHD2:

deix, 2018-11-21:
Hi,

I have seen, that there have been available some new methods to obtain polar alignment in recent PHD2 releases. Would it not be beneficial for the user (especially the older ones, that not yet have used the new methods) to have kind of "overview" about possible choices: advantages, disadvantages, requirements connected with selecting specific methods. I am thinking about the table like this could contribute to understanding the method and helping choosing the best one: (please note, that the values used are "dummys" and need to be updated)

[image]

Source: [Polar alignment.txt](https://github.com/OpenPHDGuiding/phd2/files/2604340/Polar.alignment.txt)

Best regards
Michal

agalasso, 2018-11-21:
great idea, thanks for the suggestion

bwdev01, 2019-08-23:
Fixed with commit 6f126e0

closed 2019-08-23 by bwdev01