OpenPHDGuiding/phd2 #741: BLC not adequately adapting to a certain case of stiction

agalasso, 2019-01-10:
BLC is not backing off fast enough when there is a single frame delay in the mount motion. In other words, if the first frame after the BLC lands the guide star near zero dec offset, but stiction (or some other delay) cases the overshoot to be observed in the following camera frame, then BLC does not detect the overshoot and does not scale down the BLC pulse size and the overshoot can repeat and result in oscillation.

agalasso, 2019-01-12:
update: the problem is at least partially related to a ZWO camera frame buffering problem.

```
18:59:56.047 00.000 4476 Handling exposure in thread, d=2000 o=3 r=(1102,177,31,31)
18:59:56.642 00.595 4476 Exposure complete
```
The 2000ms exposure completing in 595 milliseconds indicates that PHD2 must have received a buffered camera frame.

agalasso, 2019-01-12:
I contacted ZWO, waiting for reply.

bwdev01, 2019-01-14:
Changes made in commit 21282ed to make BLC somewhat more sensitive to stiction.  But the original problem does not appear to be a problem in the BLC.

agalasso, 2019-01-19:
after some more analysis we have a better understanding of the problem. we believe it is a result of operating the camera in video mode.  in video mode exposures start automatically before phd2 requests for them to start. by the time phd2 requests an exposure the exposure may already be nearly complete.
the result is that the completed exposure represents the position of the guide star before or during phd2's mount correction. for large corrections (like BLC or calibration steps) this causes various problems like the reported overshoot issue, but could also cause streaked star images and dropped frames.  I suspect this may be a source of excessive dropped frames during calibration.

agalasso, 2019-01-19:
more discussion in this thread: https://groups.google.com/d/topic/open-phd-guiding/rEvdN0FYkuw/discussion

agalasso, 2019-01-29:
ZWO reports that snap mode can only be used on USB3 cameras and the USB2.0-mini cameras.  Older cameras have to continue to use video mode and cannot be fixed.

commit c2460787 (2019-01-29): ZWO cameras: use snap mode for USB3 cameras and the newer USB2-mini cameras.  Fixes #741
closed 2019-01-29 by agalasso