OpenPHDGuiding/phd2 #745: PHD2 failed to start on Linux because of missing libtoupcam.so

pchev, 2019-01-24:
The compilation with the Toupcam driver now work fine on Linux.

But today I try to install phd2 on another computer and it failed to start with message:
phd2: error while loading shared libraries: libtoupcam.so: cannot open shared object file: No such file or directory

But it is a bit of a problem to install this lib with phd2 because it is already installed by other software. 
For example if you install the INDI toupcam driver it install:
lrwxrwxrwx 1 root root       15 janv. 24 10:58 /usr/lib/x86_64-linux-gnu/libtoupcam.so -> libtoupcam.so.1
lrwxrwxrwx 1 root root       19 janv. 24 10:58 /usr/lib/x86_64-linux-gnu/libtoupcam.so.1 -> libtoupcam.so.1.0.0
-rw-r--r-- 1 root root 11681408 janv. 24 10:58 /usr/lib/x86_64-linux-gnu/libtoupcam.so.1.0.0

So if we add the same in the phd2 package this will create a conflict making the package uninstallable.

Maybe we can add a dependency in the phd2 deb package to the INDI libtoupcam package but this may pose version compatibility problem. Also this package is not available in Debian or Ubuntu for now, only on the INDI ppa.

A dirty solution is to rename the lib libtoupcamphd2.so but I not really like that.

Any other idea?

agalasso, 2019-01-24:
So far we have avoided this kind of problem by getting static libs from the camera vendors.
Options so far:
(a) add dependency on INDI libtoupcam
(b) rename the lib libtoupcamphd2.so 
(c) I can ask ToupTek if they can give us static libs
(d) Another approach would be to package the libtoupcam.so files and use a phd2 wrapper script that sets LD_LIBRARY_PATH to load the appropriate (arch-dependent) bundled libtoupcam.so

pchev, 2019-01-25:
If it is not possible to get a static lib my preference is for the wrapper script with the lib in /usr/share/phd2/drivers/ , this as the advantage to fix the problem for any other camera in this case.

agalasso, 2019-01-25:
Ok, sounds like a good plan. I sent en email to Touptek requesting the static libs. If we do not hear back from them in a day or two let's go ahead with the wrapper script approach.

agalasso, 2019-01-28:
I heard back from ToupTek and they would prefer not to provide static libs. I'll commit the changes to go with the wrapper script approach.

commit 0b331d61 (2019-01-28): Linux: package camera SDK shared libraries; re-enable touptek cameras. Fixes #745
closed 2019-01-28 by agalasso