OpenPHDGuiding/phd2 #749: linux: PHD2 file locations should follow XDG spec

agalasso, 2019-02-02:
XDG Base Directory Specification https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html

$HOME/PHD2/  (log file location) should be $XDG_CACHE_HOME/phd2/ ($HOME/.cache/phd2/)
$HOME/.phd2/  (data files e.g. darks) should be $XDG_DATA_HOME/phd2/  ($HOME/.local/share/phd2)
$HOME/.PHDGuidingV2 (config file) should be in $XDG_CONFIG_HOME/phd2/config ($HOME/.config/phd2/config)
$HOME/phd2.1 (instance lock file) should be in $XDG_RUNTIME_DIR/phd2/instance-1

d33psky, 2019-03-26:
eww, but ok ;-)

agalasso, 2019-03-26:
open to other suggestions. The current behavior of dumping 4 entries into the user's home directory seems a bit rude. ;)  At least going with XDG would put the private phd2 stuff out of sight under already-existing XDG locations (~/.cache/ ~/.local/ etcetera)