OpenPHDGuiding/phd2 #757: INDI mount connection regression between v2.6.5 and v2.6.6

d33psky, 2019-03-26:
I can connect to my mount with v2.6.5 but not with v2.6.6
I'll zoom in deeper later to find the actual commit that broke things, but maybe someone has enough to go on with this ticket as is already

agalasso, 2019-03-26:
fwiw I was able to connect and use the INDI mount and camera simulators with 2.6.6 with no issues

d33psky, 2019-03-31:
I can confirm connecting to the mount simulator works.
Running the indiserver with -vvv shows that : 
`Client 0: read newSwitchVector 10micron CONNECTION`
happens on 2.6.5 and not on 2.6.6
I have not found the cause yet.

d33psky, 2019-03-31:
Zooming in deeper: with 2.6.6 when I set the port field to 'gm2000hps:3490' the connection fails. When I leave this field empty (and this use the drivers defaults config file for the 'host' and 'port') the connection succeeds.

pchev, 2019-03-31:
At the moment PHD2 only support setting a serial or usb port. 
The INDI property we use is DEVICE_PORT as defined in https://www.indilib.org/develop/developer-manual/101-standard-properties.html
To set the connection to Ethernet you cannot just set the host:port in this same field.

I look at the 10micron driver and it define this additional properties:
- CONNECTION_MODE , serial or tcp
- CONNECTION_TYPE , tcp or udp
- DEVICE_ADDRESS.ADDRESS for the telescope host name
- DEVICE_ADDRESS.PORT for the ip port

All of this must be managed by PHD2 if we want it work.
But this properties are not listed in the standard properties page, so we have no idea if another device will use the same or other to get the same functionality.
So it is probably best you not set the port in PHD2 and define all the required options in the driver settings.

Ideally this Port field must be removed to let all the driver configuration in the INDI panel but I remember that some driver do not reload the port when started so it is very convenient that PHD2 do that.
For more clarity, maybe the text in the INDI Camera Setup panel can be changed from "Port" to "Local Port"

agalasso, 2019-03-31:
Thanks, Patrick. I will change the label to "Device port" and the tool-tip text to "Device connection port" to match the terminology in the INDI docs.

commit 56b7d0d1 (2019-03-31): INDI: clarify meaning of device port connection option in the INDI device connection window. Fixes #757
closed 2019-03-31 by agalasso