OpenPHDGuiding/phd2 #759: PHD2 crashes with ZWO ASI120MM

rkaczorek, 2019-04-02:
I've been using ZWO ASI120MM for guiding for quite a long time. I flashed with compatible firmware for linux and it works quite well. Sometimes it renders bad frames but I can live with this. I have noticed recently that PHD2 crashes with this camera. It looks like it happens when it cannot take a shot from the cam.
`00:01:13.310 00.000 1945518688 Move returns status 0, amount 0
00:01:13.310 00.000 1945518688 move complete, result=0
00:01:13.310 00.000 1945518688 worker thread done servicing request
00:01:13.310 00.000 1953320960 GuideStep: -16.3 px 1434 ms EAST, -11.3 px 0 ms NORTH
00:01:13.310 00.000 1945518688 Worker thread wakes up
00:01:13.311 00.001 1945518688 worker thread servicing REQUEST_EXPOSE 2500
00:01:13.311 00.000 1945518688 Handling exposure in thread, d=2500 o=3 r=(901,300,51,51)
00:01:13.311 00.000 1945518688 ZWO: frame (896,288)+(64,64)
00:01:13.311 00.000 1945518688 ZWO: stopcapture
00:01:14.280 00.969 1945518688 ZWO: startcapture
00:01:34.315 20.035 1945518688 ZWO: getimagedata ret 11
00:01:34.316 00.001 1945518688 ZWO: stopcapture
00:01:36.785 02.469 1945518688 worker thread setting skip send exposure complete
00:01:36.785 00.000 1945518688 worker thread queueing reconnect event to GUI thread
00:01:36.785 00.000 1945518688 Error thrown from /build/phd2-gNkfQS/phd2-2.6.5.rev20190201/worker_thread.cpp:157->Capture failed
00:01:36.785 00.000 1945518688 worker thread skipping SendWorkerThreadExposeComplete`

PHD2 2.6.5.rev20190201-0ppa1~ubuntu16.04 on Raspberry Pi 3B+ running Ubuntu Mate 16.04

agalasso, 2019-04-03:
hi @rkaczorek could you please upload a full debug log from a session where you saw a crash.
log upload instructions: https://openphdguiding.org/getting-help/

agalasso, 2019-04-29:
closing this issue as we cannot do anything without the requested log file.

closed 2019-04-29 by agalasso