OpenPHDGuiding/phd2 #763: PI Zero

sinitcin, 2019-04-16:
Hello! I try to build phd2 for PI Zero. On armbian in PIZero i have are internal compiler error. But, for RPI 3 build processing passed successfully. I would like to cross compile from x64 Ubuntu 18.10, can i do it? Or can i download arm-deb package? Please help me.

agalasso, 2019-04-16:
sorry, we do not have an arm-deb package available
if there are specific compilation errors related to the PHD2 source code we can help you with those
for questions about cross-compiling you would need to check in a pi-zero support forum

agalasso, 2019-04-29:
closing this as there is nothing we can do without any info about the compilation errors, and general cross-compilation questions are better answered by asking an a Pi forum

closed 2019-04-29 by agalasso