OpenPHDGuiding/phd2 #772: Cloud simulation not working - fix suggested

mikefulb, 2019-06-20:
I have been testing some automation code I'm writing and tried to enable the cloud simulation in the PHD2 simulation camera dialog.  What I got was just a 'white noise' image.

Looking at the code around here:

https://github.com/OpenPHDGuiding/phd2/blob/54ddade315b4de026b85720cd677eded0c89f1ea/gear_simulator.cpp#L899

it appears the problem is simply that you need to add the 'cloud effect' to the exists value of the pixel, so adding '*p + ' to the start of the line would fix it.  Also might want to clamp the result of the sum with a max() so it doesn't wrap around?

agalasso, 2019-06-20:
maybe a good solution would be to replace the clouds checkbox with a cloud opacity slider, then you could simulate any amount of clouds including the two options available currently (no clouds / fully clouded-out)

mikefulb, 2019-06-20:
I made the change so it reads '*p += ' so it is adding to the current pixel value and it now draws the stars with an elevated background that depends on the clouds_inten property.

If you look at:
https://github.com/OpenPHDGuiding/phd2/blob/54ddade315b4de026b85720cd677eded0c89f1ea/gear_simulator.cpp#L906

you'll see it currently just assigned a scaled random number to the pixel which will overwrite whatever was previously assigned (stars, comets) to the rendered image.  Since hot pixels are rendered after the clouds they might still show up.

So there is an actual bug which prevents the rendering of clouds + stars to work properly in the current code.

But when modified as I explain above it tends to saturate the stars - it is adding in an offset component which is also handled elsewhere even when clouds are not requested so it seems perhaps to be doubling up on the background level and thus pushing the star peaks up to saturation.

So I agree as an enhancement there ought to be a more refined method to control the cloud strength.

I found for my testing a change to:

*p += std::min(65535, *p + (unsigned short) fabs((rand() % 15000)));

did a nice job and by setting the exposure time to different values I could control how often star lock was lost which is really what I wanted for testing my recovery code.  This also prevent wrap around on the pixel values when the sum goes over 65535.

Obviously this was just a quick hack for my use and I'm not suggesting this as a fix.

agalasso, 2019-06-20:
The original intent of the code was never to render clouds + stars, only clouds + hot pixels, as one would see with a camera in fully clouded-out conditions. The main goal was to allow testing the various schemes for dealing with hot pixels when no stars were visible.  I do like your suggestion of adding additional capability for simulating partial cloud conditions.

If we replace the clouds checkbox with a slider for cloud opacity, the clouds and stars could be combined with the usual type of opacity expression:
    `pixel_val = opacity * cloud + (1 - opacity) * stars`
where `cloud `is the current value computed in `render_clouds` and `stars` is the value computed when the clouds checkbox is un-checked, and `opacity` is the clouds opacity value from the slider (between 0.0 and 1.0)

mikefulb, 2019-06-20:
Ah I understand thanks for explaining.

mikefulb, 2019-06-21:
For my purposes I wanted to test the lost of star lock detection in my code for imaging.  I'm using a common metric of how many times the star is lost per minute to gauge if it is hopeless to continue imaging.  Here is a quick and dirty implementation that obviously needs reworking for any permanent use:

https://github.com/mikefulb/phd2/blob/fff9e5d7de24f1c8fff8591fde33ec9be589f2ed/gear_simulator.cpp#L916

My goal was to have an average number of 'star lock lost' events per minute but with some temporal variability.

If for a given rendered frame the cloud is not in effect I dialed in a slight "cloud" effect but I'm not sure that is useful.

I did another quick hack of dividing the original image with rendered stars by 3 as it gave me a good test image with a star just on the verge of  detection but that would need to be worked out in a final implementation.  I'm just linking this experiment to see if that was the direction you were thinking.

If interesting I could integrate this into the code base and add some controls for the frequency of clouds events (change the CLOUD_PROB) as well as the strength of clouds events (your opacity parameter).

PR #872 (2020-08-04): Change simulator cloud feature to use slider-controlled opacity level…
…. Fixes #772
commit 2e85c73a (2020-08-04): Change simulator cloud feature to use slider-controlled opacity level. Fixes #772
closed 2020-08-04 by agalasso