OpenPHDGuiding/phd2 #775: [API] get_current_equipment could report the equipment kind

pludov, 2019-07-15:
Proposal:

My idea would be to add a "driver" or "kind" attribute to the get_current_equipment API response, that makes it possible to know whereas a cam (or mount or anything) is available through INDI or anything else (ASCOM...)

Reason:

I am building a UI that exposes PHD2 to mobile device, using its API.
Specifically, if the cam is a INDI camera, the ui can display the image from INDI driver.

In order to do that,I find the device name using the get_current_equipment and assume it is a INDI device. But it could be other supported drivers as well.

agalasso, 2019-07-15:
PHD2 does not have a notion of driver type, but ASCOM and INDI mounts and cameras always have "ASCOM" or "INDI" in the device name presented in the UI.  The device names being shown in the guide log and the event server API were not showing "ASCOM" or "INDI", but I just fixed that in 07b3274.
Hopefully checking for "INDI" in the camera name should be sufficient for your application.  If not, we can explore it further.

closed 2019-07-15 by agalasso

pludov, 2019-07-15:
Thank you, that's fine for my need. But isn't the "name" field already used by existing client ? I mean if something is already using it it may break now.

agalasso, 2019-07-15:
yeah, that is a valid point, but I am not aware of any clients that look for specific names. The names come from the ASCOM or INDI driver, so it's even possible that the name could change with a driver update so it would not be good for PHD2 clients to look for specific driver names.

commit 07b32740 (2019-07-15): make sure INDI/ASCOM mount and camera names include "INDI" or "ASCOM". Fixes #775