OpenPHDGuiding/phd2 #776: API: invalid json sent when new line present in message

pludov, 2019-07-15:
When a warning includes a new line, the message sent over the event monitoring API contains unescaped newlines, which is not valid in JSON.

Example of data received:

```javascript
{"Event":"Alert","Timestamp":1563193992.492,"Host":"xubuntudev","Inst":1,"Msg":"After 16.0 sec the camera has not ... of the Advanced Settings Dialog.
PHD will make several attempts to re-connect the camera.","Type":"warning"}
```

Should indeed be send has:

```javascript
{"Event":"Alert","Timestamp":1563193992.492,"Host":"xubuntudev","Inst":1,"Msg":"After 16.0 sec the camera has not ... of the Advanced Settings Dialog.\nPHD will make several attempts to re-connect the camera.","Type":"warning"}
```

See: https://stackoverflow.com/questions/16690101/can-a-json-value-contain-a-multiline-string

agalasso, 2019-07-15:
I cannot reproduce this.  I am seeing the newline escaped as required.
`{"Event":"Alert","Timestamp":1563207155.865,"Host":"AGW540","Inst":1,"Msg":"After 16.0 sec the camera has not completed a 1.0 sec exposure, so it has been disconnected to prevent other problems. If you think the hardware is working correctly, you can increase the timeout period on the Camera tab of the Advanced Settings Dialog.\nPHD will make several attempts to re-connect the camera.","Type":"warning"}`

Are you using an up-to-date version of PHD2?   There was a bug in this area but it was fixed a while ago by 6f50bcaba5566345364f5bfe5d1446b4afe62be6

pludov, 2019-07-15:
Sorry, My bad ! 
It work as expected, I was just not running the version i was thinking...

closed 2019-07-15 by pludov