OpenPHDGuiding/phd2 #784: Simulator seeing

TheDiscoDuck, 2019-08-06:
In gear_simulator.cpp, sigma (line 1064 in version 2.6.6) is set as seeing_scale / (image)scale * seeing_adjustment).

seeing_adjustment is a constant. Ideally, I think sigma should go down as the exposure time increases, so I'd suggest one should modify line 1064 to something like

double sigma = SimCamParams::seeing_scale / (seeing_adjustment * SimCamParams::image_scale * sqrt(exptime / 1000));

This is essentially because the std dev of the distribution reduces as the sqrt of the number of iid distributions being averaged.

NOTE: this may necessitate a change in the empirical factor of 2.4 in the previous line, depending on what exposure time that factor was established at.

bwdev01, 2019-08-06:
I considered this at the time of the implementation and decided it wasn't worth the time needed to further calibrate the empirical terms in the algo.  I still have that opinion.  The empirical terms encapsulate a number of factors including scaling limitations in the simulator and the reality that seeing is not strictly Gaussian.   The seeing behavior in the simulator is only intended as a crude estimation, and the simulator is not really intended for exploring that sort of behavior.  It is primarily used for testing new guide algos, comparing guide algos, testing unusual types of star deflections that can't be replicated, etc.  Secondarily, it can be used by users for daytime experimentation of PHD2 and how the features work.  None of those things would benefit much by making the seeing simulation more exact.

TheDiscoDuck, 2019-08-06:
Not a problem to leave as is. Just thought I'd flag it.

I'm developing a guide algorithm that responds to seeing changes (essentially changes the aggression dynamically) and so I needed that feature for testing - but obviously I can just run with that change in my local dev version. Thanks for the response.

closed 2019-08-07 by agalasso

bwdev01, 2019-08-07:
If all you want to do is control the relative contribution of the seeing
component, you can already do that with the simulator config UI.  That
control is enabled even if guiding is active.
 
 
  _____  

 

<https://github.com/OpenPHDGuiding/phd2/issues/784?email_source=notification
s&email_token=ADDHSVZRF2YXZKWZKGKTVXLQDHZXJA5CNFSM4IJUGNV2YY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD3WT2JQ#issuecomment-518864166>
it on GitHub, or mute
HZXJANCNFSM4IJUGNVQ>  the thread.
<https://github.com/notifications/beacon/ADDHSV5ITY23SNJBKKOO4TLQDHZXJA5CNFS
M4IJUGNV2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD3WT2JQ
.gif>

TheDiscoDuck, 2019-08-07:
It's actually the combo of exptime and seeing I was playing with i.e. trying to balance the two, hence I needed the seeing to respond to the exposure time change. But that's ok. As I said, doable locally.
Thanks for the comments.