OpenPHDGuiding/phd2 #789: guiding assistant: add option to detect star saturation level

agalasso, 2019-08-24:
Star saturation detection is more reliable when the "by max ADU" option is selected, but this requires knowing the camera's saturation level adu value.
It would be nice if the G.A. had an option to measure star saturation level by taking increasingly longer exposures and looking at the max ADU value of the brightest star(s).
The saturation level could then be used to populate the saturation level ADU setting and select saturation level mode star rejection option.

bwdev01, 2019-08-24:
I'm not sure this belongs in the GA.  Seems like it's more of a one-shot
thing like the reverse-Dec-on-meridian-flip option.  And why can't we work
with the BitsPerPixel property of the cameras?
 
Bruce
 
  _____  

saturation level (#789)
 

<https://github.com/OpenPHDGuiding/phd2/issues/789?email_source=notification
s&email_token=ADDHSV65VNGLXGHFIW252I3QGFU5RA5CNFSM4IPGX6A2YY3PNVWWK3TUL52HS4
DFUVEXG43VMWVGG33NNVSW45C7NFSM4HHGEKZA>  it on GitHub, or mute
FU5RANCNFSM4IPGX6AQ>  the thread.
<https://github.com/notifications/beacon/ADDHSV4HY62WS3ML7CWKY23QGFU5RA5CNFS
M4IPGX6A2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HHGEKZA.gif>

agalasso, 2019-08-24:
> Seems like it's more of a one-shot thing like the reverse-Dec-on-meridian-flip option

sure, that is reasonable.  But seems like a good option to have at GA time since at that time we know the user is in the process of setting stuff up and has the mount tracking and pointing at a field of guide stars.

> why can't we work with the BitsPerPixel property of the cameras?

It's not 100% reliable since not all 16-bit cameras saturate at 65535. For example, I have a camera that saturates at 65000.    Also, somebody may want to set the value a little lower than the max to avoid the non-linear adu range which will vary by camera.
But perhaps we could put those issues aside and use the 8-bit or 16-bit limit which should work for most users.

bwdev01, 2019-08-25:
I'm in favor of using the simplest approach and worrying about sensor
nuances only if people encounter problems with it.  Anything we do in this
area is likely to be better than what we have.
 
  _____  

star saturation level (#789)
 

<https://github.com/OpenPHDGuiding/phd2/issues/789?email_source=notification
s&email_token=ADDHSV4XBRNI6OLO74AUQOTQGGT4VA5CNFSM4IPGX6A2YY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD5CIBRQ#issuecomment-524583110>
it on GitHub, or mute
GT4VANCNFSM4IPGX6AQ>  the thread.
<https://github.com/notifications/beacon/ADDHSV6LZAVG3MTOP7BCVUDQGGT4VA5CNFS
M4IPGX6A2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD5CIBRQ
.gif>

agalasso, 2019-08-25:
:+1:  good, let's change the default to "by ADU" and use the camera's bitsperpixel to set the initial threshold
we should also add a bitsperpixel check whenever the camera connects and update the threshold (along with invalidating the dark lib) if the bitsperpixel values changes -- several cameras (e.g. ZWO) allow the bitsperpixel to be specified by the user before connection

bwdev01, 2019-08-25:
Ok, roger that.
 
  _____