OpenPHDGuiding/phd2 #798: GearDialog::DoConnectCamera needs to call SetCoolerOn for ReconnectCamera

d33psky, 2019-09-19:
in case the cooling feature exists and was enabled.

d33psky, 2019-09-19:
See also this thread https://www.cloudynights.com/topic/673847-on-the-edge-of-defeat/

bwdev01, 2019-09-19:
This link just leads to a long lament about unresolved reliability issues
with a particular ZWO camera.  If you think there's something in there
that's relevant to the specific feature request in PHD2, could you please
summarize it.
 
Thanks,
Bruce
 
  _____  

SetCoolerOn for ReconnectCamera (#798)
 

https://www.cloudynights.com/topic/673847-on-the-edge-of-defeat/
-
<https://github.com/OpenPHDGuiding/phd2/issues/798?email_source=notification
s&email_token=ADDHSV7HN4ZMIPMPIOBJPRDQKNMHNA5CNFSM4IYJUTR2YY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7DC3HY#issuecomment-533081503>
it on GitHub, or mute
NMHNANCNFSM4IYJUTRQ>  the thread.
<https://github.com/notifications/beacon/ADDHSV4XNNHEARNXM5AIZJLQKNMHNA5CNFS
M4IYJUTR2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7DC3HY
.gif>

d33psky, 2019-09-20:
Caught it myself after 26 hours or 28 268 1s images. A ZWO: getexpstatus EXP_FAILED , after which PHD2 retried it 3 times then gave up and reconnected to the camera (and did not enable cooling) but continued making images.

After this the cooler was indeed off. I verified that physically by walking up to the camera as well.

```
01:48:08.401 00.000 140737072191232 worker thread servicing REQUEST_EXPOSE 1000
01:48:08.401 00.000 140737072191232 Handling exposure in thread, d=1000 o=3 r=(0,0,0,0)
01:48:09.702 01.301 140737072191232 ZWO: getexpstatus EXP_FAILED, retry exposure
01:48:11.004 01.302 140737072191232 ZWO: getexpstatus EXP_FAILED, retry exposure
01:48:12.305 01.301 140737072191232 ZWO: getexpstatus EXP_FAILED, giving up
01:48:12.323 00.018 140737072191232 worker thread setting skip send exposure complete
01:48:12.323 00.000 140737072191232 worker thread queueing reconnect event to GUI thread
01:48:12.323 00.000 140737072191232 Error thrown from /home/hans/src/phd2/worker_thread.cpp:157->Capture failed
01:48:12.323 00.000 140737353747008 Alert: Lost connection to camera
PHD will make several attempts to re-connect the camera.
01:48:12.323 00.000 140737072191232 worker thread skipping SendWorkerThreadExposeComplete
01:48:12.323 00.000 140737072191232 worker thread done servicing request
01:48:12.331 00.008 140737353747008 Try camera reconnect, now = 1568936892
01:48:12.331 00.000 140737353747008 gear_dialog: ReconnectCamera
01:48:12.331 00.000 140737353747008 gear_dialog: DoConnectCamera [ZWO ASI Camera]
01:48:12.331 00.000 140737353747008 Status Line: Connecting to Camera ...
01:48:12.515 00.184 140737353747008 GetString("/profile/4/cam_hash/fb6ac968ce0ca5aa/whichCamera", "") returns ""
01:48:12.515 00.000 140737353747008 Connecting to camera [ZWO ASI Camera] id = []
01:48:12.516 00.001 140737353747008 ZWO: find camera id: [], ncams = 1
01:48:13.193 00.677 140737353747008 ZWO: using mode BPP = 8
01:48:13.194 00.001 140737353747008 ZWO: usb3 = 1, is_mini = 0, name = [ZWO ASI1600MM-Cool]
01:48:13.194 00.000 140737353747008 ZWO: selecting snap mode
01:48:13.194 00.000 140737353747008 ZWO: IsColorCam = 0
01:48:13.194 00.000 140737353747008 ZWO: supported bin 0 = 1
01:48:13.194 00.000 140737353747008 ZWO: supported bin 1 = 2
01:48:13.194 00.000 140737353747008 ZWO: supported bin 2 = 3
01:48:13.194 00.000 140737353747008 ZWO: supported bin 3 = 4
01:48:13.389 00.195 140737353747008 ZWO: camera has cooler
01:48:13.389 00.000 140737353747008 ZWO: gain range = 0 .. 600
01:48:13.390 00.001 140737353747008 ZWO: lowest RN gain = 300 (50%)
01:48:13.390 00.000 140737353747008 ZWO: frame (0,0)+(4656,3520)
01:48:13.424 00.034 140737353747008 Connected Camera: ZWO ASI1600MM-Cool
01:48:13.424 00.000 140737353747008 FullSize=(4656,3520)
01:48:13.424 00.000 140737353747008 PixelSize=3.80
01:48:13.424 00.000 140737353747008 BitsPerPixel=8
01:48:13.424 00.000 140737353747008 HasGainControl=1
01:48:13.424 00.000 140737353747008 GuideCameraGain=50
01:48:13.424 00.000 140737353747008 HasShutter=0
01:48:13.425 00.001 140737353747008 HasSubFrames=1
01:48:13.425 00.000 140737353747008 ST4HasGuideOutput=1
01:48:13.425 00.000 140737353747008 GetBoolean("/profile/4/camera/AutoLoadDefectMap", 1) returns 1
01:48:13.425 00.000 140737353747008 auto-loading defect map
01:48:13.425 00.000 140737353747008 Loading defect map file /home/hans/.phd2/darks_defects/PHD2_defect_map_4.txt
01:48:13.425 00.000 140737353747008 Defect map file not found: /home/hans/.phd2/darks_defects/PHD2_defect_map_4.txt
01:48:13.425 00.000 140737353747008 Status Line: Defect map not loaded
01:48:13.426 00.001 140737353747008 GetBoolean("/profile/4/camera/AutoLoadDarks", 1) returns 1
01:48:13.426 00.000 140737353747008 Auto-loading dark library
01:48:13.426 00.000 140737353747008 Error thrown from /home/hans/src/phd2/myframe.cpp:2313->File does not exist
01:48:13.426 00.000 140737353747008 failed to load dark frames from /home/hans/.phd2/darks_defects/PHD2_dark_lib_4.fit
01:48:13.426 00.000 140737353747008 Status Line: Darks not loaded
01:48:13.427 00.001 140737353747008 Status Line: Camera Connected
01:48:13.430 00.003 140737353747008 Camera Re-connect succeeded, resume exposures
01:48:13.430 00.000 140737353747008 ScheduleExposure(1000,3,0) exposurePending=0
01:48:13.430 00.000 140737353747008 Enqueuing Expose request
01:48:13.431 00.001 140737072191232 Worker thread wakes up
01:48:13.431 00.000 140737072191232 worker thread servicing REQUEST_EXPOSE 1000
01:48:13.431 00.000 140737072191232 Handling exposure in thread, d=1000 o=3 r=(0,0,0,0)
01:48:13.431 00.000 140737072191232 ZWO: set CONTROL_EXPOSURE 1000000
01:48:15.700 02.269 140737072191232 Exposure complete
```