OpenPHDGuiding/phd2 #799: gtk gdk assertion failed: (pExtents->y1 < pExtents->y2)

d33psky, 2019-09-22:
Is below assert failure our fault or something in gtk ?
```
**
Gdk:ERROR:/build/gtk+2.0-AoeliP/gtk+2.0-2.24.32/gdk/gdkregion-generic.c:337:miSetExtents: assertion failed: (pExtents->y1 < pExtents->y2)

Thread 1 "phd2.bin" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff3b40e97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff3b42801 in __GI_abort () at abort.c:79
#2  0x00007ffff0a9a335 in g_assertion_message () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff0a9a39a in g_assertion_message_expr () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff195de2e in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#5  0x00007ffff196b39f in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#6  0x00007ffff196bfb0 in gdk_window_process_all_updates () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#7  0x00007ffff196bff9 in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#8  0x00007ffff194ac1c in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#9  0x00007ffff0a73285 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007ffff0a73650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007ffff0a73962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007ffff1d11a37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#13 0x00007ffff6fb8b05 in wxGUIEventLoop::DoRun() () at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#14 0x00007ffff696ba93 in wxEventLoopBase::Run() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#15 0x00007ffff69330a6 in wxAppConsoleBase::MainLoop() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#16 0x00007ffff69bdae9 in wxEntry(int&, wchar_t**) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#17 0x00005555557e8ece in main(int, char**) (argc=1, argv=0x7fffffffdcb8) at /home/hans/src/phd2/phd.cpp:84
```

agalasso, 2019-09-22:
We cannot tell from the backtrace whether that is related to PHD2 or if it is an internal gdk issue.
Are you able to reproduce it?
If, so what are the steps to reproduce it?   Also, what Linux distro are you using?

d33psky, 2019-09-23:
I've seen it a few times now. It reproduces hard, takes at least a night to trigger it. Does not happen every night.
I have the gdb session still open. Anything particular I could extract ?
This is on Ubuntu 18.04.3 LTS

agalasso, 2019-09-23:
what is happening in phd2 when the problem reproduces?
looping exposures? guiding? sitting idle?
which tool windows are displayed? (graph? ao graph? none? etc.)
what camera and mount are you using (attaching a debug log would be useful)

here's a related report in that seems to implicate a threading issue:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=834396

d33psky, 2019-09-23:
PHD2 was guiding at 3s exposure intervals, SX-AO was in use (built in driver), SX Lodestar camera (also built in driver), 10Micron mount via INDI, API connection from EKOS.
Tool windows in use are: AO position, Star profile, Camera view, Guide stats and History.
Everything looks fine to me at the moment the assert triggered. Here's the GUI screenshot at the point where the assert happened. I'll and the debug log next.

[image]

That gnuplot threading issue has indeed the same assertion failed listed: (pExtents->y1 < pExtents->y2) (at A1). Which thus points to GTK. Maybe I should try to run a debugging PHD2 without using a debugging GTK so that these particular asserts are not armed. But how ? And is that even a good idea ?

agalasso, 2019-09-23:
thanks for the info.  yes, it could be an internal gtk issue, or it could be that some thread in phd2 other than the main thread is somehow erroneously doing something GUI-related (thus causing a concurrency problem).  That should never happen, but who knows.
I doubt the asserts can be disabled (nor should they be).
To see if something in one of PHD2's worker threads or in one of the INDI worker threads is doing something UI-related, you could try some experiments like:
 - try using the simulator in phd2 rather then the real gear and see if the problem goes away
 - try closing all the optional display windows (graphs, star profile, etc.) and see if the problem goes away

d33psky, 2019-09-23:
Good ideas. I tried the simulator and after about 1 hour this happened : 
```
Gdk:ERROR:/build/gtk+2.0-AoeliP/gtk+2.0-2.24.32/gdk/gdkregion-generic.c:1110:miUnionNonO: assertion failed: (y1 < y2)

Thread 2 "phd2.bin" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe731d700 (LWP 17588)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff3b40e97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff3b42801 in __GI_abort () at abort.c:79
#2  0x00007ffff0a9a335 in g_assertion_message () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff0a9a39a in g_assertion_message_expr () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff195e226 in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#5  0x00007ffff195f682 in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#6  0x00007ffff195fc79 in gdk_region_union () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#7  0x00007ffff196a584 in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#8  0x00007ffff196a755 in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#9  0x00007ffff6fd5117 in wxWindow::Refresh(bool, wxRect const*) ()
    at /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#10 0x00005555557553c9 in GraphStepguiderWindow::ShowBump(PHD_Point const&) (this=0x555556709400, curBump=...)
    at /home/hans/src/phd2/graph-stepguider.cpp:186
#11 0x00005555556562b5 in StepGuider::MoveOffset(GuiderOffset*, unsigned int) (this=
    0x555556bcac00, ofs=0x7fffe731c8d8, moveOptions=13) at /home/hans/src/phd2/stepguider.cpp:1226
#12 0x0000555555856e02 in WorkerThread::HandleMove(MOVE_REQUEST*) (this=0x555556325000, req=0x7fffe731c8b8)
    at /home/hans/src/phd2/worker_thread.cpp:283
#13 0x0000555555857976 in WorkerThread::Entry() (this=0x555556325000) at /home/hans/src/phd2/worker_thread.cpp:414
#14 0x00007ffff6a7ad52 in wxThread::CallEntry() () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#15 0x00007ffff6a8263c in  () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#16 0x00007ffff46b86db in start_thread (arg=0x7fffe731d700) at pthread_create.c:463
#17 0x00007ffff3c2388f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
Not exactly the same, but it's 'A4' in that GTK bugreport from gnuplot.
Drat :/

agalasso, 2019-09-24:
yup, that backtrace shows the culprit -- the stepguider window is being updated in a worker thread, not the main UI thread, a big no-no.
I'll fix that ASAP.

commit e29bfd90 (2019-09-24): do not allow AO graph to be updated in threads other than the main UI thread. Fixes #799
closed 2019-09-24 by agalasso