OpenPHDGuiding/phd2 #802: PHD2 color selections for night viewing in Windows 10

wetzelmd, 2019-09-30:
Several users of SGP have created Windows 10 dark themes for setting dark colors for backgrounds or shades of red and amber to help preserve night vision when imaging in the field.
In Windows, the PHD2 window, title bar and dropdown menus change to dark theme colors, as do some of the buttons.  However, some buttons, such as Histogram graph scaling selections, remain gray with black text while the Clear buttons change to the Dark Theme colors.  The title bars on the tool windows remain a bright blue with white text.  Can their colors be tied to the colors used by the main window?  Also, the text displaying RA, DEC RMS, etc. remain White, while the text in the guiding algorithm settings boxes on the Histogram graph change to the Dark Theme colors.

Thanks in advance,

Mark W

wetzelmd, 2019-09-30:
Image snapshot
[image]

agalasso, 2019-10-01:
@wetzelmd   I'm curious why you closed this, seems like a ligit issue

wetzelmd, 2019-10-01:
Andy,

First, thanks for all the hard work you have done on PHD2.  As a retired controls engineer turning astrophotographer, I love it!

I am new to Github, and I fumbled to the close as I was editing my post after uploading the screenshot.

Mark

wetzelmd, 2019-10-01:
Reopened request