OpenPHDGuiding/phd2 #809: Increase max search region to 100px?

bwdev01, 2019-10-24:
Scientific user request for larger search region, needed for extremely fine image scale and repositioning of object on sensor/fiber location.  Need to be sure the larger search region limit won't have unwanted side effects.

agalasso, 2019-10-25:
A large search region can be a way for users to shoot themselves in the foot. Mainly it is an issue for guide scopes and when guiding in star-rich fields when the search region is more likely to contain multiple stars. Somebody might be tempted to make a large search region to workaround a misbehaving mount, but then get into trouble when there are multiple stars in the search region.

Maybe we could set the search region size limit based on the image scale? That would allow it to be larger (number of pixels) when the image scale is small.

bwdev01, 2019-10-25:
Yes, good idea.  That's what I meant about "unintended side effects."  For
this particular use-case, the image scale is extremely small, something like
0.2 asp.
 
  _____  

(#809)
 

<https://github.com/OpenPHDGuiding/phd2/issues/809?email_source=notification
s&email_token=ADDHSV5N6WZ7H2B3FZHDYJTQQJ7ZVA5CNFSM4JEXYQY2YY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECHHH3Q#issuecomment-546206702>
J7ZVANCNFSM4JEXYQYQ> .
<https://github.com/notifications/beacon/ADDHSVYSXBBKG3C7BZYRYZDQQJ7ZVA5CNFS
M4JEXYQY2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECHHH3Q
.gif>

d33psky, 2019-10-25:
FWIW I too am (still) at a for me ridiculous scale of 0.25 arcseconds per pixel. (F12 at 3m).  And I have increased the search region in the code to 100 often.
Making it image scale dependent sounds great.