OpenPHDGuiding/phd2 #813: Add Guide Assistant API so it can be RPC controlled.

MapleEve, 2019-11-15:
As the sky condition varies and seeing changes night by night, if it’s possible to have an API option to operate Guide Assistant?

The recommended min-motion setting help a lot to reduce the total RMS. And I have to manually run a Guide Assistant before opening the SGP to adjust its value.

It's also criteria that the PA error estimates can be reported, so the other program may give some additional info about the GA's final information.

agalasso, 2019-11-16:
@esazx, I don't think your imaging is likely to benefit from running the Guiding Assistant each night. It should be adequate to run the G.A. once on a night with average or better-than-average seeing. If seeing degrades so far that the guider is chasing the seeing more than it would have with a fresh G.A. run, then your subs are going to be low quality anyway (due to the poor seeing), and you probably won't even care about the guiding.
I'm basing this on personal experience with my own imaging setups and from looking at lots of people's guide logs.  In my case, I set my min-move values once based on a night of average seeing and after that I never touch them.
If you still want to pursue this, perhaps you could attach some guide logs to make the case?

MapleEve, 2019-11-16:
[PHD2_GuideLog_2019-10-03_172750.txt](https://github.com/OpenPHDGuiding/phd2/files/3853961/PHD2_GuideLog_2019-10-03_172750.txt)
[PHD2_GuideLog_2019-10-30_185126.txt](https://github.com/OpenPHDGuiding/phd2/files/3853962/PHD2_GuideLog_2019-10-30_185126.txt)
[PHD2_GuideLog_2019-10-29_165730.txt](https://github.com/OpenPHDGuiding/phd2/files/3853963/PHD2_GuideLog_2019-10-29_165730.txt)

These a recently new moon night guiding, I have the weather data:

Oct 3, wind 3m/s guts 7m/s, seeing 0.85 avg,
Oct 28-29  wind 5m/s guts 10m/s, seeing 0.7 avg,
Oct 28-29 wind 2m/s guts 6m/s, seeing 0.75 avg.

Settings for  Max RA/DEC duration, PE method and min-move varies, and I run the GA each night before the sequencing.

agalasso, 2019-11-16:
thanks for the data, quite helpful. I should have asked for this before, but if you have the corresponding debug logs could you attach those too (either here or upload using PHD2: Help > Upload Log Files...)

agalasso, 2019-11-17:
@MapleEve I took a quick look at your guide logs and a couple issues are apparent -- unrelated to this request or to he Guiding Assistant.
Your guiding is struggling due to the inappropriate EQMOD ASCOM default settings. Please see https://github.com/OpenPHDGuiding/phd2/wiki/EQASCOM-Settings
In particular, make sure your guide rate is _at least_ 0.5x, prefereably higher, like 0.9x.   You'll need to redo your calibration after you fix the mount's guide speed setting.
Also, in one of the sessions you switched over to the Z-Filter guide algorithm.  I don't think this is a good match for your mount. I'd recommend you switch back to Resist Switch for dec and Hysteresis for RA. Predictive PEC for RA would also be a valid choice.

MapleEve, 2019-11-18:
> @MapleEve I took a quick look at your guide logs and a couple issues are apparent -- unrelated to this request or to the Guiding Assistant.
> Your guiding is struggling due to the inappropriate EQMOD ASCOM default settings. Please see https://github.com/OpenPHDGuiding/phd2/wiki/EQASCOM-Settings
> In particular, make sure your guide rate is _at least_ 0.5x, preferably higher, like 0.9x. You'll need to redo your calibration after you fix the mount's guide speed setting.
> Also, in one of the sessions you switched over to the Z-Filter guide algorithm. I don't think this is a good match for your mount. I'd recommend you switch back to Resist Switch for dec and Hysteresis for RA. Predictive PEC for RA would also be a valid choice.

oops! That's a huge thanks to point out. It's strange for some windy night that the guide cannot running so well.

https://openphdguiding.org/logs/dl/PHD2_logs_2hKE.zip

These are all related debug logs, but missing Oct 2-3's data, I think the log is missing and I cannot find it xD

agalasso, 2019-11-27:
@MapleEve I took a look at your guide logs and indeed we do see that you are getting wildly different min-move recommendations from night to night.   The source of the variation in the GA min move recommendations was largely addressed in the newer versions of PHD2 where the guiding assistant min-move recommendation code was reworked.

You should upgrade to the latest version of PHD2 available here: https://openphdguiding.org/development-snapshots/

With the newer version of PHD2 you will get more consistent min-move recommendations from night to night.

A single guiding assistant run on a night of average seeing will give you min move settings that are valid on any night and you won't need to run the G.A. every night.   I expect the new G.A. version will give you a recommendation around 1.0 px, close to the value you got at this GA run: PHD2_DebugLog_2019-10-30_185126.txt:23:15:13.391.

Please let me know how that works out.

MapleEve, 2019-12-16:
@agalasso 
I'm trying the Dec 7th beta version these nights, and I will give some feedback after reviewed the logs.

Thanks for the great work. That's really helpful.

Maple