OpenPHDGuiding/phd2 #817: QHY cameras: update code to use new callback-based SDK functions

agalasso, 2019-11-26:
MYQ from QHY recommended we switch PHD2 to use the new callback-based methods in the QHY SDK.

agalasso, 2019-11-26:
documentation is here: https://www.qhyccd.com/file/repository/latestSoftAndDirver/SDK/Demo/ExplanationofQHYCCDSDKThreeInterfaceDevelopmentModesSampleCode.doc
see page 24

rkaczorek, 2019-12-03:
Is this related to this report? https://indilib.org/forum/astroberry/6052-astroberry-server-2-0-0-is-available-now.html?start=72#46495

agalasso, 2019-12-03:
completely unrelated

That post looks like somebody complaining the the old QHY5 interface in PHD2 does not support whatever camera they were trying to connect to.   The old QHY5 camera selection is literally only for the old QHY5 camera.  All modern QHY cameras use the selection "QHY Camera" in PHD2.  See here for info: https://github.com/OpenPHDGuiding/phd2/wiki/CameraSupport