OpenPHDGuiding/phd2 #822: detect AP ascom driver bug and give warning

agalasso, 2019-12-20:
The AP ASCOM driver can hang if the Synchronous Pulse Guide option is enabled.  PHD2 should detect when the pulse guide operation was synchronous and issue an alert.

https://groups.google.com/d/topic/open-phd-guiding/jpczS27wyek/discussion

bwdev01, 2019-12-20:
How would we know?  That setting is managed by this particular driver, I
don't think there's any ASCOM property to look at. Are you thinking we'd
look at the elapsed time between the start of the command and a return from
the call?  Then conclude that a return longer than the requested pulse-size
implies the call is synchronous?  How reliable could that be on a low-end
Windows system?
 
Bruce
 
  _____  

(#822)
 

https://groups.google.com/d/topic/open-phd-guiding/jpczS27wyek/discussion
-
<https://github.com/OpenPHDGuiding/phd2/issues/822?email_source=notification
s&email_token=ADDHSVZAX5AAS4GV5L4NW73QZUJ2ZA5CNFSM4J6BTHQKYY3PNVWWK3TUL52HS4
UJ2ZANCNFSM4J6BTHQA> .
<https://github.com/notifications/beacon/ADDHSV6N6HTRCR64B7GMXBDQZUJ2ZA5CNFS
M4J6BTHQKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4ICBLAHQ.gif>

agalasso, 2019-12-20:
the latter -- check the elapsed time to return from PulseGuide
when the sync guide option disabled Pulse returns within less than 1ms.  With the option enabled Pulse returns after the full guide duration has elapsed.   We could have some kind of "three strikes and you are out" policy to avoid false positives.
Also I think we should add an option in the brain, something like  "check for driver quirks" that can be used to disable this check and any other checks we add for other mounts.

bwdev01, 2019-12-20:
So you want to set some Boolean in Scope_ASCOM based on the literal string
returned from the driver id?  Has any of these beleaguered users just asked
AP to fix this driver bug?
 
  _____  

warning (#822)
 

<https://github.com/OpenPHDGuiding/phd2/issues/822?email_source=notification
s&email_token=ADDHSV4DLWFFLP7IEOJ2UEDQZURFHA5CNFSM4J6BTHQKYY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHOBQBY#issuecomment-568072199>
URFHANCNFSM4J6BTHQA> .
<https://github.com/notifications/beacon/ADDHSV5UCGNBUQVBGE5Y7U3QZURFHA5CNFS
M4J6BTHQKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHOBQBY
.gif>

agalasso, 2019-12-20:
right, we would have to trigger the behavior based on the driver id or name.

I totally agree this is an ugly workaround and the real solution would be for the ASCOM driver developer to fix the ascom driver.  The issue *was* reported to them when we first learned about it.

If Bryan's report is correct and the new default value is "on", then we are going to see a lot of reports of this issue. From a pragmatic point of view, we have an opportunity to make the experience a lot better for PHD2 users -- an alert message when guiding starts, rather then a hard-to-diagnose hang that users are going to naturally attribute to PHD2.  Trading a big win in user experience (and supportability) for a small bit of ugly driver-specific code in PHD2.

Loosely related issue: #808 - in that case too we need to do something specific based on the ASCOM driver id or name.   Likewise, we already do workaround a quirk in the Gemini ASCOM driver (detected by driver name.)

commit b124eb9d (2019-12-24): Add check for AP ASCOM driver sync guide pulses. Fixes #822
closed 2019-12-24 by agalasso