OpenPHDGuiding/phd2 #829: Version 2.6.6dev3 crashing on macOS when trying to upload logs

rbarbera, 2020-01-07:
To reproduce the crash:

* Open PHD2 v 2.6.6dev3 on macOS (in my case 10.14.6 Mojave)
* Go to Help > Upload Log Files...
* Select a Log run and debug info (or only the log run)
* Press Next
* Press Upload

The app crashes with the attached crash report

[PHD2Crash.txt](https://github.com/OpenPHDGuiding/phd2/files/4030226/PHD2Crash.txt)

rbarbera, 2020-01-07:
The problem remains on 2.6.6dev4

[PHD2_266_dev4_crash.txt](https://github.com/OpenPHDGuiding/phd2/files/4031129/PHD2_266_dev4_crash.txt)

agalasso, 2020-01-07:
@rbarbera thanks for letting us know.  I have been trying to reproduce it without success but I will keep trying.

rbarbera, 2020-01-07:
If you need more information, please let me know. It happens always since I installed dev3. Before that I’ve been uploaded guide logs without problems. 

Saludos,
Rafa Barberá

agalasso, 2020-01-07:
I found a reliable way to reproduce it and have traced it down to an issue in wxWidgets. We're a release behind on wxWidgets (3.1.2 => 3.1.3), so I'm going to see if it's fixed in the newer version.

Repro steps:
   do something that brings up one of the phd2 progress dialogs (I'm using TEST_SLOW_CONNECT in gear_simulator.cpp)
  wait for the progress dialog to pop up
  switch focus to some other app => phd2 crashes

agalasso, 2020-01-07:
confirmed fixed in wxWidgets 3.1.3  (also saw a note about it in the 3.1.3 ChangeLog)

starting with the next release (dev and non-dev builds), we'll build OSX 64-bit with wxWidgets 3.1.3

OSX build instructions have been updated to indicate wxWidgets 3.1.3

closed 2020-01-07 by agalasso