OpenPHDGuiding/phd2 #832: UI Scaling on high DPI screens

RononDex, 2020-01-17:
I am having issues with PHD2 scaling with high DPI screens:
[image]
This is a screenshot from my surface book (which runs Arch Linux and I use to control my astro setup).

The checkbox is incredibly tiny compared to the text (screen is 14.3", 3000x2000)

agalasso, 2020-01-17:
The size of the check-boxes is out of the control of PHD2 and is determined by Arch.
On my phd2 development machine with 4K monitor I have my display scaling setting at the default (Auto). (System Settings > Display > Settings > User interface scaling).
From your screenshot fragment it looks like you are perhaps scaling fonts? (on my system: System Settings > Font Selection > Text scaling factor).   I've got it at 1.4 on my system.

RononDex, 2020-01-21:
The size of checkboxes being controlled by Arch doesnt make any sense to me...
Arch is a distro, it has 0 influence what so ever on DPI scaling
DPI scaling comes from X11, GDK or QT

agalasso, 2020-01-21:
> DPI scaling comes from X11, GDK or QT
yes, exactly correct  (it does not come from PHD2, and PHD2 has no control over it)

RononDex, 2020-01-22:
But why do other applications just work fine then?

agalasso, 2020-01-22:
I have no idea

RononDex, 2020-01-24:
Hmm strange, I'll have to check some config files then, thanks for the help though :)

agalasso, 2020-01-24:
ok, please let us know what you find so we can help other users who encounter this problem.

I'm going to close this ticket since there is nothing actionable at this time for phd2.

closed 2020-01-24 by agalasso