OpenPHDGuiding/phd2 #835: Feature Request: Define region within which auto-select will attempt to find star

graemecoates, 2020-02-02:
Couldn't see a request for a similar feature - would it be possible to have a feature to define an area within which the auto-select routines attempt to find a star? 

I'm thinking here of a couple of use cases. One might be if you have a fairly short focal length guidescope, where it's possible to choose a guide star that's someway outside of the image area. Another similar use case might occur if you have less than perfect alignment between the guidescope and the imaging scope. 

In both cases, selection of a star in the wrong area of the autoguiding frame can result in an increased chance of field rotation - if I was able to define an area within which to accept a guide star (or at least try to find one), then this might be avoided for auto-selected guide stars. 

Thanks!

agalasso, 2020-02-02:
I'm not sure if this is useful to you or not, but region of interest for star auto-selection is already available through the server API.  https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring#available-methods
note the `roi` optional parameter to the `guide` method

graemecoates, 2020-02-03:
Ah - that's interesting to know. I'll have to check out if it's possible to pass that from SGP over to PHD then. Thanks for the pointer! (would be good to have it in the GUI as well though).

bwdev01, 2020-02-04:
Hi Graeme.  Has this been a problem you've actually encountered?  Could you
maybe give us some numerical examples where it was a practical problem?  I'm
asking only because I can't recall ever hearing of this sort of problem
being reported on the forum.
 
Thanks,
Bruce
 
 
  _____  

which auto-select will attempt to find star (#835)
 

<https://github.com/OpenPHDGuiding/phd2/issues/835?email_source=notification
s&email_token=ADDHSVYZYPZT6FYNUQVY6JDRA73ZXA5CNFSM4KOWUSDKYY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKTN5YA#issuecomment-581361376>
73ZXANCNFSM4KOWUSDA> .
<https://github.com/notifications/beacon/ADDHSV4TB5E46CSQL3TEXYLRA73ZXA5CNFS
M4KOWUSDKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKTN5YA
.gif>

graemecoates, 2020-02-04:
The example I can give is as below. [I'm fully aware that, the "correct" solution is to get better alignment, but I hope this is a reasonable example - this is fairly close to what I think I'm seeing. (another use case might be from an off axis guider that has poor enough star shapes off axis that we should restrict PHD to searching in only one area I suppose)]

Example:

* I have a dual rig, one with a moderate length refractor (say 800mm), and the other with a widefield telephoto lens based system. FoV are 1x0.7deg and 3x2deg respectively. I can alternate between using either the refractor or the telephoto system as the guider/imager. 

* Assume polar misalignment of say 4 arcmin (OK - this could probably be improved, but for a portable rig, this might be somewhere near expected). 

* Using a the telephoto as the guider for the refractor, if PHD2 picks a guide star off near a corner, then I'll get a much worse result in terms of field rotation than if it picks a star near the centre of the field. 

In theory, it could pick a guide start that is somewhere around 2.5 degrees away from the far side of the field being imaged, assuming perfect field alignment - if there's a bit more misalignment between the two systems, the result could be worse. 

With these in mind, using narrowband exposures of 20min (fairly reasonable for ensuring read noise is swamped), then the calculated field rotation distances by declination at higher decs are as follows (see: http://celestialwonders.com/tools/rotationMaxErrorCalc.html):

Dec   Rotation distance (microns)
40       4.6
50       5.5
60       7.1
70       10.4
80       20.5

If I could pin down the guide star to within the field of the refractor, then the rotation distance for furthest regions of the imaging field and the guider halves in this case above. 

Ultimately, the answer is to get better polar alignment as rotation goes fairly linearly with polar misalignment. And of course, the effect is worse at higher declination. But, I am reasonably sure I have seen this take place for a system I use for some objects in Cepheus - probably exacerbated due to a slight misalignment between guider and imager. 

Just wonder if it would be helpful, given we have the option via the server API, but not in the GUI. 

Cheers

Graeme