OpenPHDGuiding/phd2 #845: Documentation about ConfigurationChange event

gnthibault, 2020-02-28:
Dear all,

I recently realized my client was throwing error upon unknown event:
'Event': 'ConfigurationChange',

Unfortunately I haven't been able to find the corresponding description here:
https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring

What is this event used for ? What is the best way to manage it ?

Thank you in advance for your help

gnthibault, 2020-02-28:
Ah, in addition, my client is also failing, because it seems that the field for request id ("id") is not there anymore ?
This makes it difficult to acknowledge if the communication is working well or not.

Does this feature completely disappeared ?

bwdev01, 2020-02-28:
No, I think you're mistaken about this, this is a pretty critical part of
the API.  Here's a recent example showing a few command/response pairs and
you can see the id field is alive and well. :-)  These are just lines from
the PHD2 debug log file, which you can search for by using the 'evsrv' text
target.
 
23:54:02.904 00.000 7180 evsrv: cli 08E9BDF8 request:
{"method":"guide","params":[{"pixels":1,"time":10,"timeout":40},false],"id":
3}
23:54:02.920 00.000 7180 evsrv: cli 08E9BDF8 response:
{"jsonrpc":"2.0","result":0,"id":3}
 
23:54:17.839 00.312 7180 evsrv: cli 08E9B678 request:
{"method":"get_pixel_scale","id":1}
23:54:17.839 00.000 7180 evsrv: cli 08E9B678 response:
{"jsonrpc":"2.0","result":1.04757,"id":1}
 
00:25:18.159 00.000 7180 evsrv: cli 08E9BAD8 request:
{"method":"dither","params":[6,false,{"pixels":1,"time":10,"timeout":40}],"i
d":2}
00:25:18.175 00.000 7180 evsrv: cli 08E9BAD8 response:
{"jsonrpc":"2.0","result":0,"id":2}
 
The "id" field isn't present in the asynchronous event messages that you
receive - things like settling events or PHD2 state changes - because those
aren't part of an RPC.  Maybe that's where the confusion is, you have to
determine the type of message before you can know what fields it contains.
The "id" field is part of the "method" and "jsonrpc" message types.
 
Hope this helps,
Bruce
 
  _____  

event (#845)
 

<https://github.com/OpenPHDGuiding/phd2/issues/845?email_source=notification
s&email_token=ADDHSV7RW3MW7XCPQX2EYPLRFDMRLA5CNFSM4K5KZQDKYY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENH5LEY#issuecomment-592434579>
DMRLANCNFSM4K5KZQDA> .
<https://github.com/notifications/beacon/ADDHSV3B7GPRQUJHINQAG23RFDMRLA5CNFS
M4K5KZQDKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENH5LEY
.gif>

gnthibault, 2020-02-28:
Thank you, yes I saw for the "id" it was indeed my mistake, and I removed my comment few minutes after I realized that. However the ConfigurationChange event is still not clear for me.

bwdev01, 2020-03-01:
Hi there.  Could you please post your *debug* log file when one of these
events showed up?
 
Thanks,
Bruce
 
  _____  

(#845)
 

https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring
<https://github.com/OpenPHDGuiding/phd2/issues/845?email_source=notification
s&email_token=ADDHSVYG7NEDRB45H5TUPSLRFC5SFA5CNFSM4K5KZQDKYY3PNVWWK3TUL52HS4
C5SFANCNFSM4K5KZQDA> .
<https://github.com/notifications/beacon/ADDHSVZCLF2IH6H7WW2L7TDRFC5SFA5CNFS
M4K5KZQDKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IRA5GNA.gif>

agalasso, 2020-03-01:
Wiki has been updated to include the 'ConfigurationChange` event: https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring#configurationchange

closed 2020-03-01 by agalasso

bwdev01, 2020-03-01:
By now, I think you've seen the explanations for what this event message
means.  You also asked how to manage it.  Most PHD2 clients that I'm aware
of, and certainly all the ones I've written, simply ignore messages that
aren't relevant to what the application is trying to do.  The
ConfigurationChange event, for example, would be useful only if your app
needed to know whenever any guiding or configuration parameter was changed.
Event messages and supported RPCs are added to the API whenever we see a
need but there's no reason for client applications to change unless they
specifically need the new feature.
 
Bruce
 
  _____  

(#845)
 

https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring
<https://github.com/OpenPHDGuiding/phd2/issues/845?email_source=notification
<https://github.com/notifications/beacon/ADDHSVZCLF2IH6H7WW2L7TDRFC5SFA5CNFS

gnthibault, 2020-03-02:
Thank you very much !