OpenPHDGuiding/phd2 #846: save_image and get_camera_frame_size  method do not work !!!!

Greenlobby, 2020-03-02:
I tried save__image method an get_camera_frame_size method today . The two methods can not work.  
16:01:57.680 00.010 7220 evsrv: cli 2C31E238 request: {"":{"method":"save_image","id":94}}
16:01:57.690 00.010 7220 evsrv: cli 2C31E238 response: {"jsonrpc":"2.0","error":{"code":-32600,"message":"invalid request - missing method"},"id":null}

13:55:08.658 00.340 7220 evsrv: cli 2C31E238 request: {"":{"method":"get_camera_frame_size","id":92}}
13:55:08.668 00.010 7220 evsrv: cli 2C31E238 response: {"jsonrpc":"2.0","error":{"code":-32600,"message":"invalid request - missing method"},"id":null}

bwdev01, 2020-03-02:
It looks like you've mangled the JSON format of the commands, the empty
string at the start can't be parsed as a method.
 
Correct formatting: evsrv: cli 004A5ED0 request:
{"method":"get_pixel_scale","id":1}
Yours:              evsrv: cli 2C31E238 request:
{"":{"method":"get_camera_frame_size","id":92}}
 
 
 
Good luck,
Bruce
 
  _____  

do not work !!!! (#846)
 

<https://github.com/OpenPHDGuiding/phd2/issues/846?email_source=notification
s&email_token=ADDHSVZFU5PKJ6LIA5PHWU3RFN7RVA5CNFSM4K7QTWX2YY3PNVWWK3TUL52HS4
N7RVANCNFSM4K7QTWXQ> .
<https://github.com/notifications/beacon/ADDHSVYO64CZTZ3BM7DYDPDRFN7RVA5CNFS
M4K7QTWX2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IRUKURQ.gif>

Greenlobby, 2020-03-02:
Thank you very much for your reply. I feel very strange about this error. All the other methods I wrote were OK, using the same function I wrote.  I posted the running log,  please help to see why this happened, I checked  to find out  that there was no error in the sent string about save_image, but why did the server Log get this error?  I tried many times, and tried many ways , it does not work. 

 

By the way, I also encountered another problem today. In PDH method, I issued a command to take a photo by capture_single_frame, but there was no method to query when the exposure ended, so as to save the image, it might be stored incorrectly. Pls help. 

 

Thanks for your help.

 

Tha following is the PHD logs:

 

09:02:02.176 00.000 7220 PHD2 version 2.6.7 begins execution with:

09:02:02.186 00.010 7220    Windows 7 (build 7601, Service Pack 1), 64-bit edition

09:02:02.186 00.000 7220    wxWidgets 3.0.4

09:02:02.186 00.000 7220    cfitsio 3.47

09:02:02.186 00.000 7220 GetString("/frame/LogDir", "") returns "D:\SkyIntrest\PHD2 GUIDER"

09:02:02.186 00.000 7220 CoInitializeEx returns 1

09:02:02.196 00.010 7220 locale: using dir C:\Program Files (x86)\PHDGuiding2\locale exists=1

09:02:02.196 00.000 7220 GetInt("/wxLanguage", 0) returns 60

09:02:02.216 00.020 7220 locale: initialized with lang id 60 (r=1)

09:02:02.216 00.000 7220 locale: wxTranslations language set to 60

09:02:02.246 00.030 7220 GetInt("/currentProfile", 0) returns 1

09:02:02.316 00.070 7220 reset dither spiral

09:02:02.326 00.010 7220 StartWorkerThread(00000000) begins

09:02:02.336 00.010 7220 WorkerThread constructor called

09:02:02.336 00.000 7220 StartWorkerThread(00236010) ends

09:02:02.346 00.010 7220 StartWorkerThread(00000000) begins

09:02:02.346 00.000 12532 WorkerThread::Entry() begins

09:02:02.346 00.000 7220 WorkerThread constructor called

09:02:02.356 00.010 12532 worker thread CoInitializeEx returns 0

09:02:02.356 00.000 7220 StartWorkerThread(002362B0) ends

09:02:02.356 00.000 7220 GetBoolean("/ServerMode", 1) returns 1

09:02:02.366 00.010 11044 WorkerThread::Entry() begins

09:02:02.366 00.000 11044 worker thread CoInitializeEx returns 0

09:02:02.846 00.480 7220 GetInt("/profile/1/NoiseReductionMethod", 0) returns 0

09:02:02.856 00.010 7220 GetDouble("/profile/1/DitherScaleFactor", 1.000000) returns 1.000000

09:02:02.866 00.010 7220 GetBoolean("/profile/1/DitherRaOnly", 0) returns 0

09:02:02.866 00.000 7220 GetInt("/profile/1/DitherMode", 0) returns 0

[... 1476 lines cut ...]
09:15:57.681 00.010 7220 evsrv: cli 08B90368 response: {"jsonrpc":"2.0","error":{"code":2,"message":"failed to start exposure"},"id":93}

09:15:57.681 00.000 7220 evsrv: cli 08B90368 disconnect

09:15:57.681 00.000 7220 evsrv: cli 08B902C8 connect

09:15:57.691 00.010 7220 evsrv: cli 08B902C8 request: {"method":"capture_single_frame","params":[5000],"id":93}

09:15:57.691 00.000 7220 StartSingleExposure duration=5000

09:15:57.701 00.010 7220 StartSingleExposure: camera not connected

09:15:57.701 00.000 7220 evsrv: cli 08B902C8 response: {"jsonrpc":"2.0","error":{"code":2,"message":"failed to start exposure"},"id":93}

09:15:57.701 00.000 7220 evsrv: cli 08B902C8 disconnect

09:15:57.711 00.010 7220 evsrv: cli 08B90368 connect

09:15:57.711 00.000 7220 evsrv: cli 08B90368 request: {"method":"capture_single_frame","params":[5000],"id":93}

09:15:57.721 00.010 7220 StartSingleExposure duration=5000

09:15:57.721 00.000 7220 StartSingleExposure: camera not connected

09:15:57.721 00.000 7220 evsrv: cli 08B90368 response: {"jsonrpc":"2.0","error":{"code":2,"message":"failed to start exposure"},"id":93

agalasso, 2020-03-03:
@Greenlobby  please do not post log files as a giant comment. It makes the conversation unreadable (and it would be nice if you could delete your log from the above comment please.).  You can use the built-in log upload tool in PHD2 to save your log, or upload it to any file sharing site like dropbox etc. and share the link.

To answer your question about `save_image`: the method saves the current image -- the image you see when looking at the main window in PHD2.  The event notifications tell your app when a new image is available (GuideStep etc.)

Regarding your JSON formatting issues, you may want to use one of the existing client bindings:  [PHD2 API Client Code](https://github.com/agalasso/phd2client).

I'm going to close this issue as we happen to know that the messages *do* work -- willing to re-open it if you have an example of any phd2-side issue (i.e, phd2 not operating as advertised, versus a client-side issue of sending malformed messages.)

closed 2020-03-03 by agalasso

Greenlobby, 2020-03-03:
Thank you for giving me the instructions. I have deleted the inserted log lines.  

Thank you again ,I will try to do under your instructions.

 

Dr. Gan Zhiyin

Email: 5685113@qq.com

Greenlobby, 2020-03-03:
Thank you for your help. My equipment has been using PHD guiding star. What I want to  is  to use the guide camera photos directly for solving postion, because it is easy to solve by using guider image. But using method: capture a single frame requires guiderstop and stop looping, so I have no way to get the end of exposure by GuiderStep.  Would you pls help me how to  solve this problem?  

 

Thank you very much.

 

Dr. Gan Zhiyin

Email: 5685113@qq.com

agalasso, 2020-03-03:
One way to do it would be to call `capture_single_frame` then wait for the event `LoopingExposures`. The `LoopingExposures` event is sent by PHD2 at the completion of every camera exposure when phd2 is not guiding. (When phd2 is guiding, then the GuideStep is sent for every camera frame.

Another way would be to call `loop` and then you will get a `LoopingExposures` event for every camera frame.