OpenPHDGuiding/phd2 #848: Cannot guide with spectroscopic slit

kehak, 2020-03-16:
Hello,

I'm trying to use PHD2 for on-slit spectroscopic guiding.
The spectroscopic is a Lhires with a 2800mm focal length and an ASI 120MM guiding camera.

Here are two screenshots of a magv 4 and magv 6 star with the 120MM:
https://imgur.com/IZV7RSl
https://imgur.com/BD1K9tW

And here are the main settings:
https://imgur.com/iGLt6lU

I'm currently using AudeLA which has no trouble guiding on the star, but PHD2 won't let me select it, let alone guide it.

I've patched guider_onestar.cpp and star.cpp to remove some troublesome checks, but it still won't let me select the target star. Obviously there's no spare star to guide with at that focal length, so this is not an option.

Is this a known issue with PHD2? How can I make this work?

Best

agalasso, 2020-03-16:
@kehak could you upload raw fits images for us to look at (in phd2: file > save image)
we can't do much with screenshots or histogram-stretched compressed jpg images

kehak, 2020-03-16:
@agalasso Of course, here's the FITS file for the first screenshot (typical guiding session).
Let me know if you need anything else.
[guider-phd2.zip](https://github.com/OpenPHDGuiding/phd2/files/4340831/guider-phd2.zip)

bwdev01, 2020-03-17:
This object has an apparent diameter of over 100 pixels.  PHD2 isn't intended to be used on extended objects like this.  I don't do spectroscopy but my impression is that people use fainter, more typical guide stars for guiding while using the positioning tools in the app to get their target on the slit.  This means the camera needs to be well focused and I suspect yours is not.  It also looks to me like the extended object is essentially saturated but maybe that doesn't matter for your situation.

kehak, 2020-03-18:
@agalasso @bwdev01 Indeed my first sample wasn't the best. Find attached two better guiding samples (low and high magnitude) in case it could help.

> PHD2 isn't intended to be used on extended objects like this. 

This is more or less my point, is there a way this could work? I'm currently using AudeLA which guides similarly the spectroscope of the 193cm telescope at the OHP observatory. Its 29m focal length requires guiding on the target star because of the reduced FOV, and I understand this is more or less the same with others observatories. This isn't as severe with a C11 3m FL but the luxury of a separate guiding star will seldom occur.

I understand this is an edge case, but I'm very interested in the Event Monitoring as I'd like to use PHD2 as the guiding engine for a new software stack (AudeLA is impossible to containerize and modernize). I'm not a C++ developer but let me know if I can help you implement this.

[guiding-samples.zip](https://github.com/OpenPHDGuiding/phd2/files/4346319/guiding-samples.zip)

agalasso, 2020-03-18:
@kehak thanks for offering to help.  The biggest gap is that phd2's star centroid calculation is tuned to track well--focused guide stars with a PSF diameter of no more then about 6-8 pixels.
If you have an algorithm (code) that can identify the centroid of the large stars from your gear we can work with you to get it implemented in PHD2.  The code need not be C/C++, any other language would be fine as we can convert it to C++.  Perhaps we could use AudeLA's centroid identification algorithm if you think it is effective (and contingent on the AudeLA source code license allowing sharing the code). If you want to work on identifying the AudeLA centroid code, that would be a good place to start.

kehak, 2020-03-18:
@agalasso Thank you very much for this. I've started diving into its code and will get back to you when I've found the relevant algorithm. Cheers.

kehak, 2020-03-19:
@agalasso So, here's the centroid calculation algorithm, retrieved from my Git repo:
https://github.com/kehak/audela/blob/master/audela/src/audela/libaudela/src/buf_tcl.cpp#L5878

There are 3 calculation modes implemented:

cmdSlitCentro with starDetectionMode = 1  --> Gaussian fitting (default, what I'm using in these examples)
cmdSlitCentro with starDetectionMode = 2 --> on-slit detection (not used here)
cmdFiberCentro --> fiber-fed echelle spectrometers (ESO and OHP)

The default mode is the most versatile and can guide on small and large stars, even if they're not perfectly round shaped. This is what I am using and most interested in for PHP2.

If it helps, the calculation code is called from this part of the GUI:
https://github.com/kehak/audela/blob/master/audela/gui/audace/camerathread.tcl#L215

I am sorry the comments are in French, thankfully the variables aren't and should be self explicit.

Please let met know if I can help you furthermore.

kehak, 2020-03-19:
And here's the meat of it, to spare you grepping the whole repo...

https://github.com/kehak/audela/blob/master/audela/src/audela/libaudela/src/cbuffer.cpp#L2218

kehak, 2020-04-18:
Hello dear PHD2 team,

Python has a gaussian fitting algorithm that works well on my two FITS files.
A friend of mine wrote a demo from the following guide:
https://scipy-cookbook.readthedocs.io/items/FittingData.html
(chapter "Fitting a 2D gaussian")

He was successful finding the centroid of my lowmag.fits file (the hardest to find the centroid for):

[image]

Find attached the relevant Python code. Apparently there are similar functions in C++, in particular for the Scipy "optimize" function:
http://dlib.net/dlib/optimization/optimization_least_squares_abstract.h.html#solve_least_squares

Hopefully this could help you manage the slit guiding on high focal length telescopes.

Please let me know if I can help you furthermore with this.
[centroid_fits.zip](https://github.com/OpenPHDGuiding/phd2/files/4497845/centroid_fits.zip)