OpenPHDGuiding/phd2 #849: Measure backlash independent of GA

shlomiv, 2020-03-23:
Hi,

After nearly hijacking a thread on the group (https://groups.google.com/d/msg/open-phd-guiding/Zb-zhxvXLas/gawml73sBAAJ), I thought I should open this ticket.

Essentially, it would be really useful to be able to measure backlash independent of the entire GA sequence. 

When I experienced problematic backlash, I had to mess around with the mount's hardware (push-pull, etc). Every tiny change makes a difference, and to be sure I was in the right direction, I wanted to measure the backlash often. 

Currently, every backlash measurement also involves measuring seeing as well. This way it takes a minimum of 2 minutes for the backlash test to start start. I ended up eyeballing the screen and moving the mount N/S manually while counting out loud to try and measure progress. This way I could only really measure very large changes in backlash, which made the process much harder than it could have been.

Could there be an option for an independent backlash measurement? 

Thanks!

bwdev01, 2020-03-23:
The backlash test has to be part of the GA because it must include correction for Dec drift.  Without that correction, the calculation of backlash (actually reversal delay) will be inaccurate and non-repeatable.  So the 2-minute sampling requirement is there for a good reason, it normally takes that long to get enough data points to compute a linear regression that isn’t perturbed by seeing.  I think the idea of being in a hurry and using bad data is not a good approach.  But if you insist on that, you can do a ctrl-click on the ‘Cancel’ button in the “waiting” dialog and the backlash test will start.  As I said, I think this is a bad idea and you will most likely end up chasing your tail with your mechanical adjustments.

closed 2020-03-23 by bwdev01

shlomiv, 2020-03-23:
Interesting, I did not know of the strong relationship between GA and backlash measurement. 

Its not about being in a hurry, but about having a meaningful feedback loop so that adjustments could be made. I always run a proper GA+backlash once I think I am done with the adjustments.

One more option I could think of - after successful GA, could we be allowed to rerun just the backlash measurement part again? We could argue that all the required data has been gathered, and simply rerun the north/south measurement procedure again should not give us bad data. Is that assumption correct, or am I still missing something?

Thanks for your response!

shlomiv, 2020-03-23:
A comment about chasing my tail - I was not looking for 100 ms changes in backlash, I was looking at multi-seconds differences in my backlash (imagine 14 seconds to 6). I doubt those would be caused due to seeing, but admittedly, I know very little.

In any case, I am past that and my guiding is now good for my setup (thanks AP!). I was simply reminded of this terrible night by the comment on the forum.

bwdev01, 2020-03-23:
Ok, I was really wondering what you were doing.  If you were just trying to
get from "bad" to "reasonable", I think running the GA was maybe the wrong
approach.  A simpler technique would have been to start looping, then do a
qualitative backlash measurement by using the Manual Guide tool.  You'd
issue a series of north moves until you see the star is clearly moving.
Then issue another series of 1 sec or 500 ms guide pulses to the south,
keeping track of how many you do.  As soon as the guide star starts to move
in the reverse direction, you have a rough measure of the backlash.  .
Another approach you could take would be this: 1) Run the GA for 2 minutes
and get a reasonable measure of your polar alignment error and your current
backlash.  2) When you want a quick measurement again, let the GA run until
it first produces a polar alignment estimate that's fairly close to the
value you got in the first run - then do the <ctrl>-click on the 'cancel'
button at that point
 
The purpose of the backlash test is primarily to compute a value for the Dec
backlash compensation feature or to decide that the backlash compensation
mechanism wouldn't work.  That's why I'm not in favor of complicating the UI
or trying to remember the previous result in the interest of this edge case.

 
  _____  

 

7ODNANCNFSM4LSCPMZQ> .
<https://github.com/notifications/beacon/ADDHSVYVEV6FK4EQUBML4XDRI7ODNA5CNFS
M4LSCPMZ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEPXVKBQ
.gif>

shlomiv, 2020-03-27:
Thank you for taking the time to respond, it makes sense. Thank you also for maintaining such a useful and crucial piece of software for our astrophotography hobby!