OpenPHDGuiding/phd2 #850: No rule to make target [...] libqhy.a

gnthibault, 2020-03-28:
Problem with commit: 3d8532d812586a8a9c6f7367553fdb1ed83cadc7

Looks like the build doesn't work anymore (I use 32bit debian for arm)

make[2]: *** No rule to make target '/home/rock/projects/phd2/cameras/qhyccdlibs/linux/armv7/libqhy.a', needed by 'phd2.bin'.  Stop.

Any idea how I can get this lib ?
Thank you in advance for your help

agalasso, 2020-03-29:
that looks like your Makefiles may be out of date, you can regenerate them by re-running cmake (https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux).

The QHY SDK lib is in the source tree (https://github.com/OpenPHDGuiding/phd2/tree/master/cameras/qhyccdlibs/linux/armv7/)
but qhy changed it to "libqhyccd.a", no longer "libqhy.a".   Your makefiles are still referring to the old name, so you should be able to run cmake to regenerate the Makefiles and fix the problem.
To be sure nothing else is stale, I would be conservative and start from scratch with a clean tmp/ directory.

d33psky, 2020-03-29:
I had the same (but on x64). I did rerun cmake which did not help. Neither did make clean. A new cmake in a new directory worked fine though. I did not dig deeper as my issue was solved.

RononDex, 2020-05-21:
Had the same issue, a short `rm -rf` of the git folder + recloning the repo and recompiling fixed it