OpenPHDGuiding/phd2 #851: Scope::CheckCalibrationDuration overriding calibration step size and user cannot override it

agalasso, 2020-03-29:
https://groups.google.com/d/topic/open-phd-guiding/dUm9RYr-Umc/discussion
```
20:19:29.951 00.000 281472911574448 CalDuration adjusted at start of calibration from 1200 to 50 because of  mount guide speed  change
```
debug logs is here: https://openpphdguiding.org/save_logs/PHD2_logs_2XMS.zip

agalasso, 2020-03-29:
@bwdev01 any idea how to fix this one?

bwdev01, 2020-03-29:
> overriding calibration step size and user cannot override it (#851)

Well, the underlying problem is a buggy INDI driver for Celestron.  It's
reporting a mount guide speed of 0.208 degrees/sec or 749 arc-sec/sec.  In
the first calibration he did, the implied mount guide speeds were around
0.7x sidereal.  A few possibilities come to mind: 1) add some sanity
checking and only do the adjustment if the reported mount guide speed is
sensible or 2) remove the checking altogether for changes in mount guide
speed.  The adjustment based on binning change is probably ok.  This stuff
works ok when the driver doesn't lie and to some extent, it's kind of "extra
credit."  We could let users sort it out for themselves if they change mount
guide speeds without also adjusting the cal step-size.  I'm open to
suggestion on it.
 
Bruce
 
  _____  

3LKHANCNFSM4LV24Z7A> .
<https://github.com/notifications/beacon/ADDHSV3V3UFBPST7ZPCWSJLRJ3LKHA5CNFS
M4LV24Z7KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEQMCCCQ
.gif>

agalasso, 2020-03-29:
I think a sanity check on the rate, something like treating any rate > 2x sidereal as invalid (as if the rate is unknown) would fit the bill.

We should also report this to the INDI developers.  Do you have a log snippet or something like that we could give them that demonstrates the problem?

pchev, 2020-03-29:
Bruce, be careful that INDI do not return the guide speed in deg/sec like ASCOM but it is x sidereal. 
So to this is 0.208 x sidereal = 3 arcsec / sec.

agalasso, 2020-03-29:
Thanks Patick, I think it really must be a bug in the Celestron INDI driver since PHD2 appears to be correctly converting the INDI rate (sidereal multiple) to ASCOM rate units (degrees per second) https://github.com/OpenPHDGuiding/phd2/blob/master/scope_indi.cpp#L811

pchev, 2020-03-29:
Right, I not remembered this part.

I try the Celestron driver has it include a simulation mode. 
I think the problem is it report the value as "Guide rate % sidereal" so the value is 100x too big. 
For 0.5 x sidereal it report:
Celestron GPS.GUIDE_RATE.GUIDE_RATE_WE=50
Celestron GPS.GUIDE_RATE.GUIDE_RATE_NS=50

This correspond to the 749 "/sec Bruce see in the log.

The telescope simulator and other driver give the value as:
Telescope Simulator.GUIDE_RATE.GUIDE_RATE_WE=0.5
Telescope Simulator.GUIDE_RATE.GUIDE_RATE_NS=0.5

agalasso, 2020-03-29:
aha! good find.  Will you be submitting the INDI driver issue? If not, I can do it.

bwdev01, 2020-03-29:
This looks like the Wild West.  Celestron reports a bad result, EQMOD gets
the right result, Losmandy and iOptron report "unknown".  That's just from
scanning my hard drive for INDI mount data which is limited because I don't
generally look at problems on Linux systems.  I think the best approach is
to weed these things out in the adjustment code and only handle things that
make sense.
 
Bruce
 
  _____  

 
623HANCNFSM4LV24Z7A> .
<https://github.com/notifications/beacon/ADDHSV6TSNY6WEPCZYDVHOLRJ623HA5CNFS
M4LV24Z7KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEQNETSI
.gif>

commit 1c4c0a6a (2020-03-30): Handle bogus mount guide speeds in CheckCalibrationDuration.
Fixes #851
closed 2020-03-30 by bwdev01