OpenPHDGuiding/phd2 #854: QHY cameras: add option to operate camera in 12-bit mode

agalasso, 2020-04-12:
The QHY SDK provides `SetQHYCCDBitsMode`, but the documentation is not clear (at least to me) how to find out the valid values for the "bits" parameter, and when the API method should be called (once after connect?).  We should contact QHY for clarification.

MasterPIC, 2022-01-06:
Hi,
I gave no look at PHD source. Anyway I think
SetQHYCCDBitsMode(camhandle,16)
is required to set 16bit mode.
This instruction according to the documentation can be executed anytime, with camera still connected, before taking a new exposure...

I hope it helps
CS
Armando

daleghent, 2024-10-10:
It is not possible to operate the QHY cameras in the sensor-native bit mode. The only modes available are 8, 16, and 32 depending on camera type.

bwdev01, 2025-04-20:
8 or 16-bit modes are supported in current releases.

closed 2025-04-20 by bwdev01