OpenPHDGuiding/phd2 #857: Feature request: Ability to use "master/slave" stars for guiding

pessorrusso, 2020-05-27:
Request: Would it be possible to have a master/slave star for guiding ?

Reason: Some times the guiding start is lost (clouds/bad seeing/etc) but there are others stars to be used.

Suggested behavior:
1- Master star is always used for guiding.
2- If Master star is lost then Slave star is used for guiding
3- If Slate start is lost while using the Master star then a new Slave star is selected and guiding is unaffected as it is still using the Master star.

bvalente22, 2020-05-27:
Hi Pessorrusso

There are two things I think you are asking about, and i think both are
addressed in current version of PHD

1. Resilience when guidestar is lost - If PHD loses a guidestar, it will
attempt to reacquire a guidestar already, so there is no need for a 'slave
star'. The 'slave star' approach could also backfire if you wanted to
reacquire the original (and probably preferred star) or if there was a
better one

2. multi-star guiding - although you didn't ask about this directly, it
kind of hints at the idea of more than one star is used for guiding. there
has been a lot of discussion on this and the net result has been multi-star
guiding has not (yet?) proven to be significantly better than single star
guiding

pessorrusso, 2020-05-27:
Hi bvalente, thank you for your reply.
PHD is an amazing tool but I do see that sometimes the star used for guiding is covered by clouds then guiding goes off resulting in waste of subs/time. At the same moment that the guiding star is lost I do see that there are other star options that could be used for guiding as "second option".

I read a lot of threads about the "two star guiding" and it is clear for me that it is complex and not worth it but the master/slave approach seems simple and it would add (in my opinion) more stability to the guiding during bad seeing.

bwdev01, 2021-05-16:
Most of this has been addressed with multi-star guiding in the latest development builds.  Cloud recovery has been discussed innumerable times, it is the responsibility of the imaging app to do that.  The typical guider field of view is small, so trying to deal with "partial cloudiness" seems futile - it is inherently a short-term situation.

closed 2021-05-16 by bwdev01