OpenPHDGuiding/phd2 #858: Question: How to implement communication with self-made mount?

AlfTetzlaff, 2020-06-13:
Hi everybody!
I have equipped an old mount I like with some new, self-made electronics based on an arduino, so that it can be used for auto-guiding. I also wrote the auto-guider software (Linux) which captures the images and sends the velocity with which the mount should track (calculated by a PI controller). Obviously my homebrew software is far less sophisticated than phd2 and now I want to take advantage of your guide algorithms etc. The question is how to proceed with that:

1) Should I extend phd2 to support my mount in some way? Or:
2) Should I adapt my arduino code in a way that emulates some OEM mount?

I would opt for 2), but couldn't find any info on concrete implementations; which bytes should I listen for and which should I send? Maybe you can show me the shortest path and point me to the right resources. Thanks! :-)

bwdev01, 2020-06-13:
Hi Alf.  I think the best approach by far would be to implement an ASCOM (or
INDI driver) for your mount.  That will enable most of the telescope control
apps to deal with your mount - PHD2, imaging apps, planetarium apps, etc.
All mounts have proprietary implementations at the low level so you need an
intermediary standard between the applications and your mount firmware -
that's ASCOM or INDI.  The ASCOM and INDI standards for telescope drivers
enumerate a set of functions and their parameters, the ones that will be
used by applications.  The job of the driver author is to implement some or
all of those functions and communicate with the mount to get the job done.
The communication with the mount can use any mechanism and protocol you want
- that's hidden from the applications.
 
https://indilib.org/about/discover-indi.html
https://ascom-standards.org/
 
Good luck,
Bruce
 
  _____  

self-made mount? (#858)
 

NTSNANCNFSM4N46WYPA> .
<https://github.com/notifications/beacon/ADDHSV2QA43UXL5LUOO5WGDRWNTSNA5CNFS
M4N46WYPKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4JQJM6HQ.gif>

agalasso, 2020-06-13:
exactly, what Bruce said.
I'll close this issue now.

closed 2020-06-13 by agalasso

AlfTetzlaff, 2020-06-16:
Thank you for answering my question! The issue can be closed indeed. One last thing: Which standard (ASCOM or Indi) would you recommend off the top of your head?

d33psky, 2020-06-16:
That depends on what operating system you intend to use your driver on.
Windows only ? -> ASCOM
INDI is cross-platform, but in practice most drivers are intended for Linux. But can be accessed over the network by anything.
There is a 3rd option : ALPACA, but PHD2 does not support it yet (to my knowledge).

AlfTetzlaff, 2020-06-16:
Thanks! I will try to implement the thing in INDI then; the interface looks clean and the Linux support is better (which is what I need). ASCOM still seems to be too Windows focused and is ruled out if ALPACA is not supported by phd2. Furthermore the development documentation was easier to find and read for INDI, and I dont need VisualBasic etc. etc.

agalasso, 2020-06-16:
:dart: