OpenPHDGuiding/phd2 #860: display calibration info automatically after problematic calibration

agalasso, 2020-06-18:
from Bryan:
https://groups.google.com/d/msg/open-phd-guiding/Cfgh4I3fVXc/bnPSKVt5AwAJ

2) You said that Error Messages can confuse and/or annoy, and I agree, for instance the Orthogonality message.

I see many posts where the Bad Cal messages have obviously been ignored, and many hours spent trying to guide.

So would you consider instead of "Review Calibration Data" being an optional feature, actually show a sort of "Calibration Review Data" display straight after a Calibration, but only after a Calibration that has "failed" ? With a "Good" Calibration for comparison, like this ? (Apologies for the low res).
This might avoid further fruitless activity ? (and reduce the posts :->  )

bwdev01, 2020-07-03:
After further thinking,  I still don’t think this would be helpful.  The basic problem is that many people don’t want to read any of the diagnostic messages, they don’t want to click on the ‘Help’ button, they don’t want to make an explicit decision to retain or repeat the calibration or even to say they don’t want to see the message again.  So pushing graphs in front of them isn’t going to help them and in all likelihood will just be more confusing.  What would you show them if there were too few calibration steps?  What about RA and Dec guide rates that don’t make any sense?

closed 2020-07-03 by bwdev01

d33psky, 2020-07-03:
FWIW I understand that people do not want to read diagnostic messages but I also think that just showing the graphical calibration data is a much easier to understand and stronger signal than the brief text about bad calibration data.