OpenPHDGuiding/phd2 #864: RaspberryPi4: Libusb failed to open camera QHY5

crusty-dave, 2020-07-04:
I just built an image on my new RaspberryPi 4, however, I cannot connect my QHY5-P-II-m camera.

Using lsusb I can see the USB device getting created.

If I connect my ZWO290MC, I get the following message:
"PHD2 could not connect to the camera so you may want to deal with that
later. In the meantime, you can just enter the pixel-size manually along
with the focal length and binning levels.

I wonder if there are issues with the new USB devices in the RPi4?

Thanks in advance,
-Dave

agalasso, 2020-07-04:
@crusty-dave take a look in the PHD2 debug log, or upload it to our server and we can look at it https://openphdguiding.org/getting-help/
there will be some info there about why the camera connection failed.

crusty-dave, 2020-07-04:
The upload doesn't work, it looks like it tries to bring up a dialog, but I cannot find it.  However, I found the location, here is what seems to be the relevant part:

```
00:29:50.175 06.230 3008507920 CameraFactory(QHY 5)
00:29:50.176 00.001 3008507920 GetBoolean("/profile/2/camera/UseSubframes", 0) returns 0
00:29:50.178 00.002 3008507920 GetInt("/profile/2/camera/ReadDelay", 150) returns 150
00:29:50.179 00.001 3008507920 GetInt("/profile/2/camera/gain", 95) returns 95
00:29:50.179 00.000 3008507920 GetInt("/profile/2/camera/TimeoutMs", 15000) returns 15000
00:29:50.179 00.000 3008507920 GetInt("/profile/2/camera/SaturationADU", 0) returns 0
00:29:50.179 00.000 3008507920 GetBoolean("/profile/2/camera/SaturationByADU", 1) returns 1
00:29:50.180 00.001 3008507920 GetDouble("/profile/2/camera/pixelsize", 0.000000) returns 0.000000
00:29:50.180 00.000 3008507920 GetInt("/profile/2/camera/binning", 1) returns 1
00:29:50.188 00.008 3008507920 wxMessageBoxProxy(Libusb failed to open camera QHY5.)
00:30:19.493 29.305 3008507920 CameraFactory(QHY Camera)
00:30:19.494 00.001 3008507920 GetBoolean("/profile/2/camera/UseSubframes", 0) returns 0
00:30:19.495 00.001 3008507920 GetInt("/profile/2/camera/ReadDelay", 150) returns 150
00:30:19.495 00.000 3008507920 GetInt("/profile/2/camera/gain", 95) returns 95
00:30:19.496 00.001 3008507920 GetInt("/profile/2/camera/TimeoutMs", 15000) returns 15000
00:30:19.496 00.000 3008507920 GetInt("/profile/2/camera/SaturationADU", 0) returns 0
00:30:19.496 00.000 3008507920 GetBoolean("/profile/2/camera/SaturationByADU", 1) returns 1
00:30:19.496 00.000 3008507920 GetDouble("/profile/2/camera/pixelsize", 0.000000) returns 0.000000
00:30:19.496 00.000 3008507920 GetInt("/profile/2/camera/binning", 1) returns 1
00:30:19.498 00.002 3008507920 QHYCCD: SDK Version V20191115_0
00:30:19.508 00.010 3008507920 QHY: found 0 cameras
00:30:19.508 00.000 3008507920 Alert: No compatible QHY cameras found
00:31:03.722 44.214 3008507920 CameraFactory(QHY 5)
00:31:03.723 00.001 3008507920 GetBoolean("/profile/2/camera/UseSubframes", 0) returns 0
00:31:03.723 00.000 3008507920 GetInt("/profile/2/camera/ReadDelay", 150) returns 150
00:31:03.724 00.001 3008507920 GetInt("/profile/2/camera/gain", 95) returns 95
00:31:03.724 00.000 3008507920 GetInt("/profile/2/camera/TimeoutMs", 15000) returns 15000
00:31:03.724 00.000 3008507920 GetInt("/profile/2/camera/SaturationADU", 0) returns 0
00:31:03.725 00.001 3008507920 GetBoolean("/profile/2/camera/SaturationByADU", 1) returns 1
00:31:03.725 00.000 3008507920 GetDouble("/profile/2/camera/pixelsize", 0.000000) returns 0.000000
00:31:03.725 00.000 3008507920 GetInt("/profile/2/camera/binning", 1) returns 1
00:31:03.726 00.001 3008507920 wxMessageBoxProxy(Libusb failed to open camera QHY5.)
00:32:02.515 58.789 3008507920 GetString("/profile/2/name", "") returns ".!temp!profile!name!~"
00:32:02.515 00.000 3008507920 GetString("/profile/1/name", "") returns "My Equipment"
00:32:02.516 00.001 3008507920 Delete profile .!temp!profile!name!~
00:32:02.517 00.001 3008507920 GetString("/profile/1/name", "") returns "My Equipment"
00:32:02.517 00.000 3008507920 GetString("/profile/2/name", "") returns ".!temp!profile!name!~"
00:32:02.533 00.016 3008507920 GetBoolean("/Update/enabled", 0) returns 0
00:32:02.533 00.000 3008507920 GetInt("/Update/series", 0) returns 0

...

01:36:19.321 06.375 3008393232 CameraFactory(ZWO ASI Camera)
01:36:19.322 00.001 3008393232 GetBoolean("/profile/2/camera/UseSubframes", 0) returns 0
01:36:19.322 00.000 3008393232 GetInt("/profile/2/camera/ReadDelay", 150) returns 150
01:36:19.322 00.000 3008393232 GetInt("/profile/2/camera/gain", 95) returns 95
01:36:19.323 00.001 3008393232 GetInt("/profile/2/camera/TimeoutMs", 15000) returns 15000
01:36:19.323 00.000 3008393232 GetInt("/profile/2/camera/SaturationADU", 0) returns 0
01:36:19.323 00.000 3008393232 GetBoolean("/profile/2/camera/SaturationByADU", 1) returns 1
01:36:19.323 00.000 3008393232 GetDouble("/profile/2/camera/pixelsize", 0.000000) returns 0.000000
01:36:19.324 00.001 3008393232 GetInt("/profile/2/camera/binning", 1) returns 1
01:36:19.326 00.002 3008393232 GetInt("/profile/2/camera/ZWO/bpp", 8) returns 8
01:36:19.328 00.002 3008393232 ZWO: SDK Version = [1, 14, 1108]
01:36:19.333 00.005 3008393232 ZWO: find camera id: [], ncams = 1
01:36:19.352 00.019 3008393232 ASIOpenCamera ret 5
01:36:19.352 00.000 3008393232 Alert: Failed to open ZWO ASI Camera.

```

crusty-dave, 2020-07-04:
```
Bus 002 Device 002: ID 03c3:290a  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1618:0920  
Bus 001 Device 008: ID 046d:c408 Logitech, Inc. Marble Mouse (4-button)
Bus 001 Device 006: ID 045e:00db Microsoft Corp. Natural Ergonomic Keyboard 4000 V1.0
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb -v

Bus 002 Device 002: ID 03c3:290a  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x03c3 
  idProduct          0x290a 
  bcdDevice            0.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              512mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15

Bus 001 Device 005: ID 1618:0920  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
[... 215 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

```

agalasso, 2020-07-04:
@crusty-dave 

> `01:36:19.352 00.019 3008393232 ASIOpenCamera ret 5`

status code 5 (from the ASI SDK) is this:

```
ASI_ERROR_CAMERA_REMOVED, //failed to find the camera, maybe the camera has been removed
```

so IOW the SDK is failing to open the camera.

We are planning on updating PHD2 soon (in version 2.6.9dev1) to use a newer version ASI SDK. I'm not sure if that will make a difference.

Another option would be to see if the ZWO INDI driver works for accessing the camera -- the INDI driver may be using a different version of the ZWO SDK.

You may be able to get some help from ZWO tech support, perhaps they have some way to get the SDK to recognize the camera.

Since this is an issue with the SDK, not PHD2 per se, I am going to close the issue as there is nothing actionable for us at this time on the PHD2 side.

closed 2020-07-04 by agalasso