OpenPHDGuiding/phd2 #865: GuideAlgorithmZFilter memory leak

MapMaker73, 2020-07-05:
On Windows 7, VS2013, Visual Leak Detector has flagged a memory leak in the GuideAlgorithmZFilter logic (running PHD2 version 2.6.8 downloaded from GitHub July 2, or version 2.6.9 downloaded July 5, 2020). 

GuideAlgorithmZFilter::BuildFilter() creates multiple instances of the m_pFactory attribute on the heap (once when called by SetMinMove and another when called by SetExpFactor). However, no other logic deletes that heap memory at at any time. Thus both Windows 10 and Windows 7 flag memory leaks at program exit if the Z Filter algorithm has been set.

I successfuly tried two solutions to fix the memory leak:
1) Clean up the memory in the GuideAlgorithmZFilter destructor

GuideAlgorithmZFilter::GuideAlgorithmZFilter(Mount *pMount, GuideAxis axis)
    : GuideAlgorithm(pMount, axis)
    , m_pFactory(nullptr) // Ensure m_pFactory is initialized
{
    m_expFactor = DefaultExpFactor;
    m_design = BESSEL;
    m_order = 4;
    double minMove = pConfig->Profile.GetDouble(GetConfigPath() + "/minMove", DefaultMinMove);
    SetMinMove(minMove);
    double expFactor = pConfig->Profile.GetDouble(GetConfigPath() + "/expFactor", DefaultExpFactor);
    SetExpFactor(expFactor);

    reset();
}

GuideAlgorithmZFilter::~GuideAlgorithmZFilter(void)
{
   // Clean up m_pFactory   memory if it has been created
    if (m_pFactory != nullptr)
    {
        delete m_pFactory;
        m_pFactory = nullptr;
    }
}
...
bool GuideAlgorithmZFilter::BuildFilter()
{
    bool bError = false;

    try
    {
        Debug.Write(wxString::Format("GuideAlgorithmZFilter::order=%d, expFactor=%lf\n",
            m_order, m_expFactor));
        if (m_order < 1)
        {
            throw ERROR_INFO("invalid order");
        }
        if (m_expFactor < 1.0)
        {
            throw ERROR_INFO("invalid expFactor");
        }

        double corner = m_expFactor * 4.0;
        FILTER_DESIGN design = (corner < 6.0) ? BUTTERWORTH : m_design;

        m_xcoeff.clear();
        m_ycoeff.clear();
        // Before creating a new instance, clean up m_pFactory memory if it has been created
        if (m_pFactory != nullptr)
        {
            delete m_pFactory;
            m_pFactory = nullptr;
        }
        m_pFactory = new ZFilterFactory(design, m_order, corner);
        m_order = m_pFactory->order();
        m_gain = m_pFactory->gain();
        m_xcoeff = m_pFactory->xcoeffs;
        m_ycoeff = m_pFactory->ycoeffs;
...

2) Use the simpler, and more safe memory allocation via std::unique_ptr

a) I added a line to phd.h file:
#include <memory> // For std::unique_ptr

b) I then changed guide_algorithm_zfilter.h:

class GuideAlgorithmZFilter : public GuideAlgorithm
{
    FILTER_DESIGN m_design;
    std::vector<double> m_xv, m_yv;  // Historical values up to m_order
    std::vector<double> m_xcoeff, m_ycoeff;
    int m_order;
    double m_gain;
    double m_minMove;
    double m_sumCorr; // Sum of all corrections issued
    double m_expFactor;

    //ZFilterFactory *m_pFactory;
    std::unique_ptr<ZFilterFactory> m_pFactory;

protected:
...

c) and then changed guide_algorithm_zfilter.cpp:
GuideAlgorithmZFilter::GuideAlgorithmZFilter(Mount *pMount, GuideAxis axis)
    : GuideAlgorithm(pMount, axis)
    , m_pFactory()
{
    m_expFactor = DefaultExpFactor;
    m_design = BESSEL;
    m_order = 4;
    double minMove = pConfig->Profile.GetDouble(GetConfigPath() + "/minMove", DefaultMinMove);
    SetMinMove(minMove);
    double expFactor = pConfig->Profile.GetDouble(GetConfigPath() + "/expFactor", DefaultExpFactor);
    SetExpFactor(expFactor);

    reset();
}

GuideAlgorithmZFilter::~GuideAlgorithmZFilter(void)
{
}

...

bool GuideAlgorithmZFilter::BuildFilter()
{
    bool bError = false;

    try
    {
        Debug.Write(wxString::Format("GuideAlgorithmZFilter::order=%d, expFactor=%lf\n",
            m_order, m_expFactor));
        if (m_order < 1)
        {
            throw ERROR_INFO("invalid order");
        }
        if (m_expFactor < 1.0)
        {
            throw ERROR_INFO("invalid expFactor");
        }

        double corner = m_expFactor * 4.0;
        FILTER_DESIGN design = (corner < 6.0) ? BUTTERWORTH : m_design;

        m_xcoeff.clear();
        m_ycoeff.clear();
       // Properly clean up any previous m_pFactory instance on the heap before creating a new instance
        m_pFactory.reset(new ZFilterFactory(design, m_order, corner)); 
        m_order = m_pFactory->order();
        m_gain = m_pFactory->gain();
        m_xcoeff = m_pFactory->xcoeffs;
        m_ycoeff = m_pFactory->ycoeffs;

With ether of these two potential fixes, neither Windows 10 nor Windows 7 reported a zFilter logic memory leak.
Regards,
Jim

MapMaker73, 2020-07-05:
Meant to type:
a) I added a line to phd.h file:
#include <memory> // For std::unique_ptr

agalasso, 2020-07-05:
@MapMaker73 thanks for the report! we'll fix that ASAP
If you want to take a crack at it yourself, feel free to [submit a PR](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests)
If you are going to work on a PR, please assign the issue to yourself so that we know you are working on it and we don't duplicate the effort.

commit 537470ec (2020-07-06): fix mem leak in zfilter guide algo. Fixes #865
closed 2020-07-06 by agalasso