OpenPHDGuiding/phd2 #873: I have an idea about.

AstroAir, 2020-08-02:


AstroAir, 2020-08-02:
我想在网页上绘制导星曲线，但是我无法获取导星曲线的绘制方式，如果使用phd2client则只能获得rms-ra 和rms-dec，但是好想和phd2软件中导星曲线的变化情况不相符，能不能提供一些帮助，谢谢🙏
PS:I am from China ,I feel sorry that I can only speak a little English.

agalasso, 2020-08-02:
your web application could connect to PHD2 and receive each data point -- the [GuideStep](https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring#guidestep) notification messages -- including RA and Dec offset for each camera frame. You could plot the data as it arrives.  
Which phd2client are you using? one of these: https://github.com/agalasso/phd2client ?

AstroAir, 2020-08-02:
我不知道该如何读取GuideStep的信息，我用的是https://github.com/agalasso/phd2client中的c++

AstroAir, 2020-08-02:
@agalasso 我尝试了你指的那个库，但是仍然不行，只能获得rms和peak，无法有效的绘制导星曲线。接着我查看了guider.h但是发现好像没有定义GuideStep这一项，因此无法使用。我又检查了guider.cpp里面有许多的代码，但是我并没有找到有关GuideStep的代码，我该怎么办？

AstroAir, 2020-08-03:
我仔细的阅读了API手册，发现PHD2是以json格式传输数据的，所以说，我是否可以使用PHP curl与PHD2通信，并获取相应的数据？

AstroAir, 2020-08-03:
我还仔细的检查了https://github.com/agalasso/phd2client c++部分的guider.h但是我没有发现有关GuideStep的信息，这该怎么办？谢谢🙏

agalasso, 2020-08-03:
> 我仔细的阅读了API手册，发现PHD2是以json格式传输数据的，所以说，我是否可以使用PHP curl与PHD2通信，并获取相应的数据？

@AstroAir sure, you could use php to get the GuideStep messages. Connect to TCP port 4400 and read line by line.  Each line will be a JSON message, and the GuideStep message is the one you need for plotting a graph. https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring

agalasso, 2020-08-03:
> 我还仔细的检查了https://github.com/agalasso/phd2client c++部分的guider.h但是我没有发现有关GuideStep的信息，这该怎么办？谢谢pray

@AstroAir I added an enhancement request to the phd2client github repo: https://github.com/agalasso/phd2client/issues/3 

I am going to close this issue in the PHD2 repo since there's nothing we need to do on the PHD2 side.

closed 2020-08-03 by agalasso

AstroAir, 2020-08-03:
@agalasso Thanks for your help.